dsharlet/LiveSPICE #45: Post-Processing an Audio File

colejohnson66, 2015-09-20:
It would be nice to be able to post-process an audio file. That would help with effects when the computer isn't fast enough for real-time processing. Build the circuit at the highest settings (or allow the user to specify), run an audio file through the circuit, and save output to a file (depends on #44).

colejohnson66, 2015-09-20:
I haven't looked at how the audio processing works, but if the code reads directly from the input, an abstraction class that allows arbitrary input would need to be added (similar to how [System.IO.Stream](https://msdn.microsoft.com/en-us/library/system.io.stream.aspx) can be pretty much anything from a [file](https://msdn.microsoft.com/en-us/library/system.io.filestream.aspx) or [chunk of memory](https://msdn.microsoft.com/en-us/library/system.io.memorystream.aspx) to [data compressed with DEFLATE](https://msdn.microsoft.com/en-us/library/system.io.compression.deflatestream.aspx)).

alexbv16, 2015-11-07:
eventually, I'd be interested in that feature too.

dsharlet, 2020-11-05:
Sorry for the delayed reply. I think now that the next release will contain a VST plugin, I think that is the best way to cover these needs.

closed 2020-11-05 by dsharlet