dsharlet/LiveSPICE #48: Great stuff! Still actively developed!

alfonso73, 2015-12-23:
Hi dsharlet!
Congrats for this great thing here. is still actively developed?
many thanks and all the best

dsharlet, 2015-12-28:
Unfortunately, I'm not really spending much time on this anymore. My apologies for dumping it out there. 

I had some intention of developing a few new features (the one I wanted most was VST plugin generation), but I started a new job that left me with less time, and this is a side project. I was just recently thinking about new ways to build VST plugins with this, and have some concrete ideas of how to do it. I hope to get to it sometime, but can't make any promises...

alfonso73, 2015-12-28:
hi dillon,
thanks for the info and I hope you'll be able to find time for this great
project! all the best and thanks again!
a.

dsharlet, 2015-12-28:
Thank you for your kind words!

dsharlet, 2020-10-30:
I've started working on this again for now, so I guess this issue should be closed? :)

closed 2020-10-30 by dsharlet