dsharlet/LiveSPICE #51: Linux port ?

YruamaLairba, 2016-09-19:
What do think about to have a linux port ? you can probably use [mono](http://www.mono-project.com/) for the .NET code, and you can implement a [Jack](http://www.jackaudio.org/) Client to replace ASIO for this platform.

dsharlet, 2016-09-23:
Unfortunately, the big missing dependency is WPF, the framework used for the GUI (see http://www.mono-project.com/docs/gui/wpf/).

YruamaLairba, 2016-09-23:
Too bad. the other solution is to use a multiplatform GUI framework, but this would be a huge work :(

nybbs2003, 2018-02-18:
how about running in headless mode with JACK in Linux?

z26, 2018-10-10:
I'd be interested in nybbs2003's idea as well.

risicle, 2020-10-29:
The key to WPF on linux is Wine: https://ccifra.github.io/PortingWPFAppsToLinux/Overview.html

dsharlet, 2020-10-29:
The other thing I've been seeing while picking up this project again is Xamarin forms seem similar to WPF. I don't know if they are similar *enough* to make porting easy...

Another risk is this project really needs this method to have a good implementation: https://docs.microsoft.com/en-us/dotnet/api/system.linq.expressions.expression-1.compile?view=netcore-3.1#System_Linq_Expressions_Expression_1_Compile

I don't have any reason to think it's bad on Mono, but I was surprised by how good it was in .net. Compiled native code is only a little bit faster!

dsharlet, 2022-03-14:
@jopdorp very nice work, thanks! However, I think it's working by adding an executable main to Circuit, and changing that project to an executable instead of a library, which I think will break the existing users of that library (command line tests, LiveSPICE GUI, VST plugin). If you can get the CircuitTests working (basically a slightly fancier version of your CircuitSimple user of Simulation) without changing Circuit to an executable, that's something we could merge to master.

It seems like if you were to do that, basically the only change remaining would be to change the target framework and a few deps, which would be a reasonable change, although I've had a lot of issues pop up in random places when changing target frameworks (maybe WPF doesn't work with the .net5.0 target or something? I can't remember).

To caveat this for anyone else reading this issue: I think this still doesn't get the LiveSPICE GUI or interactive simulation working on Linux. Perhaps this is close to getting the VST plugin working on linux DAWs? At least that is not such a huge GUI to get working with another framework if WPF doesn't work, but it does depend on @mikeoliphant's VST hosting system, which may also be non-trivial to port to Linux.

mikeoliphant, 2022-03-14:
For Linux, it would probably make more sense to implement an LV2 plugin. I've thought about doing an LV2 implementation for AudioPlugSharp, but haven't had a personal use case for it yet.

jopdorp, 2023-05-10:
I got it to run on wine with the caffe 7.20 runner in bottles. I just had to add dotnet 6, and there was a bug with my intel graphics card, but it runs well using nvidia.

I'm using pulseaudio, windows audio and pulseaudio so there is like 100ms latency.
This should in theory be possible to solve using wineasio and jack, which are available in kxstudio, which also provides a version of wine, which might be worth trying.

dsharlet, 2023-05-11:
Really cool! Did you need to make any changes to the code to enable that to work?

jopdorp, 2023-05-29:
> Really cool! Did you need to make any changes to the code to enable that to work?

No changes to the code were needed

RustoMCSpit, 2023-07-15:
if this gets ported to linux flatpak and snap support would be cool

RustoMCSpit, 2023-07-15:
> For Linux, it would probably make more sense to implement an LV2 plugin. I've thought about doing an LV2 implementation for AudioPlugSharp, but haven't had a personal use case for it yet.

what about CLAP?

dsharlet, 2024-03-11:
Sorry it's been so long without any real closure on this issue. I've decided to consolidate the cross platform porting to a discussion here: https://github.com/dsharlet/LiveSPICE/discussions/222

closed 2024-03-11 by dsharlet