dsharlet/LiveSPICE #53: Impressed!

sonorak, 2016-10-05:
Hi dsharlet, First of all congratulation for this project, it's really innovative, like a miracle for audio plugin developing, so i hope you will find the time to keep it alive, did you ever think to integrate it with the JUCE framework? maybe generating the code of the simulated circuit, so it would be a universal tool for whoever want or directly build a vst or who want instead to generate the code and continue the project integrating other functionality, i wish you the best, and thx for this.

alfonso73, 2016-10-05:
yeah! that would be great!

dsharlet, 2016-10-05:
Would it be useful to just generate a C function? Something with the following API:

    struct FilterState {
        ...
    };

    int Process(
        FilterState *state, 
        float param0, float param1, ...,
        const float *input0, const float *input1, ...,
        float *output0, float *output1, ..., int sample_count);

I've been thinking about building a native code generator for a very long time. It seems like just dumping out a C function that implements the filter might do the job and it would be a lot easier to implement.

alfonso73, 2016-10-05:
It could be a good idea. then it would be a matter of transfering the parameter stuff and the audio buffer to the relevant VST SDK parts.

alfonso73, 2016-10-05:
Or a C++ class?

sonorak, 2016-10-05:
My idea is this, first i explain my frustration story in joining vst developing (i'm a novice in dsp but programmer for years, not professionally, it's just a passion and i have fun), i'm a musician, and as said i like programming, so it's natural, i want to build some digital effect vst, but material to study so hard to find, tutorials, very few, very hard and frustrating, so i realize there are a couple of software like Flowstone for easy to make vst, but no x64 version vst, so useless in 2016, then finally i take my time and i want to go a little deep on it no matter what and i finally find out through various source the real procedure, so first i need a circuit simulation, most used i think is LTspice, then i need to simulate the schematic, eventually test the sound through wav input and output (very slow) and eventually solve the mathematical equation that i get in C++ to be able to create a digital simulation, hard but at least it start to make sense and i know what i have to study, then i realize that exist a framework JUCE, and i discover that is what created 80% of the plugins that i use  today, seems became a standard in vst developing, it take care already of most of the thing, you just need to build your effect, and it has several API and function ready and simple a bit like the .NET framework semplicity, now i realize that there is LiveSPICE, one of the most genius ideas in all of this, so, what if LiveSPICE, over then eventually create directly a vst, convert the schematic in a prototype of JUCE code? so who want to continue a schematic created with LiveSPICE could continue expanding it using JUCE, i see a big potential for this integration to become a standard in vst development.

sonorak, 2016-10-06:
I just want to write a little update, considering that to continue expanding an existent plugin in JUCE you can simply host your plugin, probably there is not a real strong need that it is builded in JUCE, so i think that if LiveSPICE one day can export Vst x86 and x64 that's all you need, i just want to thank you again dsharlet for this and i hope you will find the time to continue it, this is a too exciting project, all the best! :)