dsharlet/LiveSPICE #54: Tube heaters, 5Y3GT, adding componets

DeRivVR, 2016-10-05:
Hi, liveSpice is amazing and would really like to be able to do more with it. I would like to be able to build tube rectifiers but without heater connections or tube options such as a 5Y3GT I don't think it's possible. Is there a way to add components?

ekometti, 2017-08-29:
I would also like to know how to do this with other types of triodes or pentodes.

krupax, 2021-03-07:
I've also been building up to making oscillators / wein bridge filter type things but most of them require pentodes which aren't currently in the installed library - how can I add models / where can I find compatible ones?

This application is incredible, I can mess about with electronics and load them into ableton without burning my fingers or electrocuting myself!