dsharlet/LiveSPICE #57: Live Spice Crashes

pcardoso73, 2016-12-14:
Hello,
I am trying Live Spice for the first time. I have installed ASIO4all and did the first example (RC filter).
My problems:
1-AUSIO4ALL recognizes HD Audio CD in Capture 1 and in capture 2 (I don't even have a CD on my PC). The outputs are HD audio speaker 1,2,3 ... Which one should I choose ?
2-After choosing any of them, when running the simulation, the program crashes.
3 - Where can I find the log file ?

The images below show the configuration I have. Is everything set properly  ?

[image]

[image]

[image]

[image]

Best regards,
Pedro

dsharlet, 2020-10-25:
My apologies for the delay in responding, and for the bulk edit of this issue. I've pushed a few significant changes that hopefully address some audio IO issues:
- Rewrote things to not use NativeUtils.dll, so there is no native code dependency. I think this might have been the culprit of most of these issues.
- Enabled the windows audio driver. This may be a useful fallback if ASIO is not working.
- Fixed a stability issue with the windows audio driver, if for some reason it was visible before.

closed 2020-10-25 by dsharlet

pcardoso73, 2020-10-26:
Thanks for replying. I will try your software again.
Thanks again.
Cheers,
Pedro

dsharlet, 2020-10-26:
Note that I haven't made a new release yet with these fixes, so unless you build it from source, you'll probably see the same behavior as last time.