dsharlet/LiveSPICE #62: No Transistor

complainer46973, 2017-09-26:
Please add transistors to the parts library! Most guitar effects pedals use one in the circuit.
Thanks.

harrismirza, 2017-10-17:
Transistors are already present, both BJT's and JFET's

alfiq, 2019-07-11:
I thinks he is talking about adding more predefined spice models to the lib, so the transistor have not to be modeled by the user, as that can be pretty challenging.

dsharlet, 2020-10-25:
Sorry for the delayed reply. LiveSPICE can read SPICE files for transistor models, so if you have a SPICE library, you can put it in the (documents)\LiveSPICE\Components folder and LiveSPICE will populate the parts library with supported models. See Readme.txt in that folder for more information.

dsharlet, 2020-11-05:
In the next release, I've added a small library of transistors and diodes, that hopefully covers the common cases. Between this and SPICE library compatibility, I think this is fixed.

closed 2020-11-05 by dsharlet