dsharlet/LiveSPICE #63: Schematic is not visible

QQQQRRRRRTTTTT, 2018-01-21:
So something happend or I clicked on something and I can no longer see the schematic. The file is till open because if i click on simulate the schematic appears i the simulation window. I tried to uninstall and reinstall the program, its still the same. I cant create a new schemtic either because i cant se it.

phorensic, 2019-12-29:
I have the same exact problem.

krupax, 2021-03-09:
This just happened to me as well - not sure if I uninstalled something else that led to it as I didn't run livespice yesterday while cleaning up my system, but first thing this morning went to try a new diagram I'd found last night and even the component window has no content. I've uninstalled and reinstalled to no effect - I'll try and clean out the registry to see if that helps next...