dsharlet/LiveSPICE #65: Failed to find partition initial conditions, simulation may be unstable.

Mikoool, 2018-01-31:
Maybe this is not issue at all. Could you please tell, what that means?
Or how to avoid this message? I use Tube (12ax7) circuit simulation,

dsharlet, 2018-01-31:
This means that we could not solve for the steady state conditions of the circuit. This is sometimes necessary because starting the simulation when the circuit is not in a steady state can lead the simulation to be unable to converge to a solution.

I've mostly seen this occur when the circuit is somehow malformed (i.e. a floating node, a short circuit, etc.). Does the circuit successfully run despite this message? If so, that suggests the circuit is valid, but tripping up the initial conditions solver somehow. I'd be interested in taking a look at it either way.

Mikoool, 2018-02-01:
Dear dsharlet,

Thanks for reply. Circuit looks ok, but also I found if capacitor used in series in signal flow, then no simulation happens at all.
Please see attached schematic file in this message.
[scheme.zip](https://github.com/dsharlet/LiveSPICE/files/1684788/scheme.zip)

dsharlet, 2020-10-29:
I looked again at this circuit (sorry for the delay). I noticed there are a few resistors that are in milliohms, when they were probably intended to be megaohms... Aside from that, the circuit appears to be solving and simulating OK for me. I do see that there is no circuit output, but I suspect there is a bug in the circuit rather than the simulation. I think to debug it, you'd need to break it down into smaller circuits so you can test them. But to be honest, I can't actually design and debug my own circuits very well :)

closed 2020-10-29 by dsharlet