dsharlet/LiveSPICE #67: Running in headless mode with JACK in Linux?

nybbs2003, 2018-02-18:


jopdorp, 2021-03-11:
I would love to be able to run this on linux with arm, I'm working on the MisTer fpga project, which runs on de10 nano. The audio needs improving and I'd like to o run a real time spice simulation on the dual core arm chipd, then write the audio output to the fpga through a serial interface

mikeoliphant, 2021-03-11:
It might be possible. While the UI is Windows-only, the core simulation code should be runnable on .NET Core under Linux. You would still need to a way to interface with JACK, though.

Alternatively, it should be possible to generate C/C++ code for the simulation and integrate that.

jopdorp, 2021-03-11:
I'm gonna have a go at trying to create a project under Linux, see what I can get to compile, starting with the Circuit directory.

what exactly did you mean by "Alternatively, it should be possible to generate C/C++ code for the simulation and integrate that."

How would I generate c/c++ code for the simulation?

mikeoliphant, 2021-03-11:
The result of the circuit analysis is a set of algebraic expressions. It should be possible to take that set of expressions and generate C/C++ code from them.

jopdorp, 2021-03-12:
I created a sln in Circuit with a main file that does a live simulation. 
It looks terrible for now because I added everything that was needed from outside of the circuit project to Simulate.cs (the main class), but it compiles and runs. I haven't really tested it yet because I still need to implement a serial input and serial output. I'm not actually planning to implement jack, because I will not be using this with the normal linux audio system, because that doesn't exist on my target project: https://github.com/MiSTer-devel/Main_MiSTer

mikeoliphant, 2021-03-12:
You also might want to look at the simulation code from the VST implementation:

https://github.com/dsharlet/LiveSPICE/blob/master/LiveSPICEVst/SimulationProcessor.cs

jopdorp, 2021-03-13:
thanks for the tip!

jopdorp, 2021-03-16:
Hi, I've implemented a much simpler simulate class. Right now it fail during the linq compilation phase.

The class: 
https://github.com/jopdorp/LiveSPICE/blob/linux/Circuit/SimulateSimple.cs

The error:

```
$ dotnet run "../Tests/Circuits/Rectifier.schx"
Circuit.TransientSolution
Unhandled exception. ComputerAlgebra.UnresolvedName: Unresolved name 'Call to unknown function 'V1'.'.
   at ComputerAlgebra.UnknownFunction.Call(IEnumerable`1 Args) in /home/jegor/development/LiveSPICE/ComputerAlgebra/ComputerAlgebra/Expression/Functions/UnknownFunction.cs:line 20
   at ComputerAlgebra.LinqCompiler.CompileExtension.CodeGenFunction(Function f, Module Module) in /home/jegor/development/LiveSPICE/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExtension.cs:line 23
   at ComputerAlgebra.LinqCompiler.CompileExtension.Compile(Function This, Module Module) in /home/jegor/development/LiveSPICE/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExtension.cs:line 34
   at ComputerAlgebra.LinqCompiler.Module.Compile(Function f, Type[] ArgTypes) in /home/jegor/development/LiveSPICE/ComputerAlgebra/ComputerAlgebra/LinqCompiler/Module.cs:line 112
   at ComputerAlgebra.LinqCompiler.CompileExpression.VisitCall(Call C) in /home/jegor/development/LiveSPICE/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExpression.cs:line 114
   at ComputerAlgebra.ExpressionVisitor`1.Visit(Expression E) in /home/jegor/development/LiveSPICE/ComputerAlgebra/ComputerAlgebra/Expression/Visitors/Visitor.cs:line 27
   at ComputerAlgebra.LinqCompiler.CompileExpression.Visit(Expression E) in /home/jegor/development/LiveSPICE/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExpression.cs:line 24
   at ComputerAlgebra.LinqCompiler.CodeGen.Compile(Expression Expr) in /home/jegor/development/LiveSPICE/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CodeGen.cs:line 162
   at ComputerAlgebra.LinqCompiler.CodeGen.DeclInit(Scope Scope, Expression Expr, Expression Init) in /home/jegor/development/LiveSPICE/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CodeGen.cs:line 289
   at ComputerAlgebra.LinqCompiler.CodeGen.DeclInit(Expression Expr, Expression Init) in /home/jegor/development/LiveSPICE/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CodeGen.cs:line 297
   at Circuit.Simulation.<>c__DisplayClass44_1.<DefineProcess>b__5() in /home/jegor/development/LiveSPICE/Circuit/Simulation/Simulation.cs:line 303
   at ComputerAlgebra.LinqCompiler.CodeGen.<>c__DisplayClass71_0.<DoWhile>b__0(Expression x, Expression y) in /home/jegor/development/LiveSPICE/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CodeGen.cs:line 471
   at ComputerAlgebra.LinqCompiler.CodeGen.DoWhile(Action`2 Body, Expression Condition) in /home/jegor/development/LiveSPICE/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CodeGen.cs:line 459
   at ComputerAlgebra.LinqCompiler.CodeGen.DoWhile(Action Body, Expression Condition) in /home/jegor/development/LiveSPICE/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CodeGen.cs:line 471
   at Circuit.Simulation.<>c__DisplayClass44_0.<DefineProcess>b__4() in /home/jegor/development/LiveSPICE/Circuit/Simulation/Simulation.cs:line 286
   at ComputerAlgebra.LinqCompiler.CodeGen.<>c__DisplayClass62_0.<For>b__0(Expression x, Expression y) in /home/jegor/development/LiveSPICE/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CodeGen.cs:line 368
   at ComputerAlgebra.LinqCompiler.CodeGen.For(Action Init, Expression Condition, Action Step, Action`2 Body) in /home/jegor/development/LiveSPICE/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CodeGen.cs:line 355
   at ComputerAlgebra.LinqCompiler.CodeGen.For(Action Init, Expression Condition, Action Step, Action Body) in /home/jegor/development/LiveSPICE/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CodeGen.cs:line 368
   at Circuit.Simulation.DefineProcess() in /home/jegor/development/LiveSPICE/Circuit/Simulation/Simulation.cs:line 252
   at Circuit.Simulation.Run(Int32 N, IEnumerable`1 Input, IEnumerable`1 Output) in /home/jegor/development/LiveSPICE/Circuit/Simulation/Simulation.cs:line 159
   at Circuit.SimulateSimple.Main(String[] args) in /home/jegor/development/LiveSPICE/Circuit/SimulateSimple.cs:line 18
```

jopdorp, 2021-03-16:
It might be that i didn't do LoadLibraries, but it is in Controls. I'm thinking V1 was meant as an interactive element, the one where you basically have to select the mocrophone

jopdorp, 2021-03-16:
I think this is the constructor that needs to be called with a set of libraries where one includes the "V1" library
```
        public Module(IEnumerable<Type> Libraries)
        {
            if (Libraries == null)
                libraries = new Type[] { typeof(StandardMath) };
            else
                libraries = Libraries.ToArray();
        }
```

mikeoliphant, 2021-03-16:
I think the problem is that you aren't hooking up the circuit's inputs and outputs.  You can have a look at the VST SimulationProcessor for how to do that.

jopdorp, 2021-03-16:
Yeah, I think you're right

jopdorp, 2021-03-16:
Diving a bit deeper into the code, I think I found the place where I could target a different execution environment.
Looking at the following method:
https://github.com/jopdorp/LiveSPICE/blob/b795a951a5a7495cc39ddf8ac2988ac9d2664c4d/Circuit/Simulation/Simulation.cs#L192
it seems like it first compiles to linq, and then in compiles linq to binary.
I'm thinking to make an abstraction layer between this to add a different target than linq.
So basically I would just add an implementation that would target Verilog(hardware design language) instead of linq for every LinqExpr.* method.
Then I'd just output the resulting verilog to a file, and leave the compilation to verdor specific tools like altera

jopdorp, 2022-03-09:
I did actually get this to run on linux:
https://github.com/jopdorp/LiveSPICE/tree/linux