dsharlet/LiveSPICE #75: Component qiestion

willbart, 2019-10-13:
Hi
Great program, to run audio through spice software is excellent!

A few questions re components, I'm a novice so forgive if they're a bit basic;
1: Cant find a polarised capacitor; Or would a diode & capacitor do the job?

2: I'm looking for silicon diodes (I want to try a diode bridge limiter) - would normal diodes behave in a similar way?

3: Is it possible to simulate a photocell?

Finally, this is a bit daft - I just wasted an hour trying to solve "Circuit has no elements" & "simulation converged at..." errors; turns out I'd forgotten the ground! (late night yesterday)
I couldn't find anything online re these errors so maybe just a "Circuit has no Ground" message?

Thanks :)

dsharlet, 2019-10-23:
Hi will, glad you were able to get it to somewhat work for you :)

1, 2: I don't know about the behavior of specific kinds of components myself. In order to determine this, you would need to look at the data sheets and theoretical models for the components you are interested in, and see how well they are modeled by the components in LiveSPICE. The generic components let you populate the model parameters yourself, so if you can find these parameters on a datasheet, you could enter them that way. 

3. It may be possible to simulate a photocell if it matches the model of another component. However, there will not be a way to simulate light in the simulation environment :)

A better error message in the case of no ground is definitely a reasonable request. Unfortunately, I don't spend a lot of time on this project anymore, but it would make a lot of sense to add this when I get back to it.

willbart, 2019-10-30:
Hi Dillon

Thanks for getting back, sorry for my delay in responding.  From your comment ("...get it to work somewhat...") it sounds like it didn't work as well as you hoped!

I guess with the computer memory you'd need to simulate complex circuits with digital audio it starts getting impractical pretty quickly - in LTSpice just running a 0.1s simulation can take some time.

I was trying to get my head around transistor/preamp biasing - thought it really useful for experimenting with resistor values - being able to make adjustments while it was running, and hear the effects was really handy!

I'll have a go with the product spec sheets etc.  Thanks for the advice.

Kind Regards

Will

  1.  It may be possible to simulate a photocell if it matches the model of another component. However, there will not be a way to simulate light in the simulation environment :)

closed 2020-10-29 by dsharlet