dsharlet/LiveSPICE #76: Help please. - Preview 12

kgreene1983, 2020-05-07:
Hi, 

Just download and setup. My sound card has an ASIO driver, but if I use it the software crashes. So using asio4all, which seems to be working in test. 

So tried to run some examples of files and i can't see any simulations. I have this info:

building circuit...
Build: 0 errors, 0 warnings
Building solution for h=2.83446712018141E-06
Performing steady state analysis...
Performing transient analysis...
System solved, 3 solution sets for 57 unknowns.

dsharlet, 2020-10-25:
There isn't a way to show the frequency response currently, and I don't really have plans to add this. I think it would require a whole new kind of user interface.

dsharlet, 2020-11-05:
I think now that the next release will contain a VST plugin, I think that is the best way to cover this need. There are many VST plugins with all kinds of analyzers.

closed 2020-11-05 by dsharlet