dsharlet/LiveSPICE #77: Exact copy of circuit, but still no output

TwistedGlass, 2020-05-22:
Whenever I try to recreate the circuit of the MXR Distortion +, I never get any output when simulating. The things that I think are possible issues:
1. Op-amp. I'm not sure if the op-amp has to be configured a certain way for each circuit but whenever I drop in an op-amp it always looks right but no sound. All of the connections are right
2. 1 mΩ potentiometers/resistors never seem to give any output with ANY circuit I've made, but that's just part of the MXR's schematic
It's not an issue with my interface. I have read multiple schematics for the MXR and have matched every single component to a T, and there's just no output.

dsharlet, 2020-10-25:
Sorry for the delay. Are you saying that the example MXR Distortion + circuit works, but when you re-create it, it doesn't?

If you are saying it looks right, but there is no sound, do you mean there is visible signals in the oscilloscope? If so, maybe this is just an audio problem (there are many people with audio problems...), do other circuits work?

The example does have a 1MOhm variable resistor in it, and it seems to be working for me.

dsharlet, 2020-10-29:
One thing I just noticed in your first message... it's possible that circuits with 1 milliohm and other components with large resistances are unstable. Having resistors (or any other kind of component) with such a huge difference in resistance might produce really wonky intermediate values during simulation. Do you really want a resistor with 1 milliohm and not 1 megaohm?

Sorry again for the delay, I'm going to close this, please reopen if you try again and my previous suggestions don't help.

closed 2020-10-29 by dsharlet