dsharlet/LiveSPICE #79: not live at all, or even crashing

phil123456, 2020-06-16:
Hello,

I do the basic rc filter tutorial, then I simulate it, 

- it's everything but live
it simulates, like any other regular spice and says simulation is done, but I can't make it run live like it is advertised to do so

- on my second sound card, it's simply crashing

any help on this ? tell me if you need more infos

thanks for the time you'll spend on this question

https://i.imgur.com/Kbywecd.png

phil123456, 2020-06-20:
is this project dead ?

OzClarke, 2020-07-04:
Hi
The last few days I've been playing around with LiveSpice and initially ran into the same issue as phil.
There's a little problem with NativeUtil.dll.  From what I can tell the current installer is using an older/maybe incomplete version of the file. I've downloaded the source code and found it was compiling to the wrong folder.

It's pretty cool once you get it working.

dsharlet, 2020-10-25:
My apologies for the delay in responding, and for the bulk edit of this issue. I've pushed a few significant changes that hopefully address some audio IO issues:
- Rewrote things to not use NativeUtils.dll, so there is no native code dependency. I think this might have been the culprit of most of these issues.
- Enabled the windows audio driver. This may be a useful fallback if ASIO is not working.
- Fixed a stability issue with the windows audio driver, if for some reason it was visible before.

closed 2020-10-25 by dsharlet