dsharlet/LiveSPICE #92: Feature Request - MIDI Input

alfonso73, 2020-11-03:
Hi! 
I'm so excited that Livespice development is active again!
Given that there is some work "under the hood" to make a VST version (super happy for this!!!),
wouldn't be great to get MIDI input to simulate and test analog synth schematics?
thanks and all the best!

mikeoliphant, 2020-11-03:
That's a really good idea. It would certainly be possible to take incoming midi notes and convert them to an input voltage signal.

dsharlet, 2020-11-03:
To clarify, the idea here would be to convert MIDI to a digital audio signal, and then pass that into the VST plugin?

mikeoliphant, 2020-11-03:
The idea (I think) would be to have the VST plugin accept MIDI input and turn that into a non-audio input signal to control circuit behavior. The actual audio would be generated by one or more oscillators in the circuit.

Something like what is described here:

https://www.allaboutcircuits.com/projects/diy-synth-series-vco/

dsharlet, 2020-11-03:
Oh I see, very interesting. This seems very cool, though the details aren't clear to me yet. Right now the only way to get data into a circuit is via a voltage/current source, defined either by a function (e.g. `Vin[t] = Sin[t*2*pi*440]`) or by translating a sequence of samples into a voltage... is that enough to support this use case?

Maybe some small refactoring would enable voltage sources to be functions of the input signal. The only other thing I can think of is supporting scalar parameters as inputs to the simulation, which could be used in voltage/current source expressions. So you could write `Vin[t] = t > X` to define a step function for example.

mikeoliphant, 2020-11-03:
Probably the simplest thing initially would be just to have the VST generate a sequence of samples.

alfonso73, 2020-11-03:
Hi @dsharlet and @mikeoliphant.
I confess that i forgot if Livespice has paramters (for pots, selecttors, switches and the likes) that can be moved in realtime to change the properties of the schematic.
Ideally both a VST Effect and VST Instrument version would have VST parameters automatable from the host.
As for MIDI i guess that Note Number, Velocity, Aftertouch, Pitch-bend and Control Changes data should be mapped to just linear CV. Probably Note Number to exponential CV given that pitch is expressed in exponential way. But in the analog synthesizer a linear CV input is transformed in an exponential CV via the exponential converter such as... https://www.schmitzbits.de/expo_tutorial/
So definitely is a matter to talk a little bit about ;-)
To get the "traditional" analog workflow probably MIDI Note Number should be converted to linear CV and then inside the LiveSpice schematic there has to be an exponential converter circuit.

dsharlet, 2020-11-08:
> I confess that i forgot if Livespice has parameters (for pots, selecttors, switches and the likes) that can be moved in real time to change the properties of the schematic.

It does have this ability, but it's a bit limited. It has to rebuild the circuit when a pot/switch changes, which takes a bit of time, so pots don't react smoothly to input.

If the signal coming from MIDI is just a sequence of samples that you can interpret as a voltage or current in the circuit, it won't need to rebuild the circuit for that of course.

It seems like it would be possible to make the 'key' map to a linear voltage level in the VST plugin, and then you could build the exponential conversion circuit Mike linked. You could also "cheat" and just do the exponentiation as an expression of the input signal inside LiveSPICE. It looks like the exponential conversion circuit would take a fair amount of complexity, and you might prefer to spend the complexity/compute power on better signal generators instead...?

zvookin, 2020-11-09:
Analog synths work on control voltage (CV), which will often be paired with a gate signal to allow keyboard control. Plenty of detail here: https://en.wikipedia.org/wiki/CV/gate . Plug-in APIs, such as VST and LV2, typically allow both digital sample based input and MIDI. MIDI is the best bet for a lower sample rate event stream. Thus building support into the VST wrapper to turn MIDI inputs into CV+Gate signals is likely pretty useful. (Implementing it from scratch using strictly analog elements is a lot of work.) Ideally the element would allow configuration of voltage ranges and mapping of channel aftertouch and all the MPE/MIDI2 features into multiple CV signals or possibly some method of internally combining some of them into one CV.

The place to start looking is probably VCVRack's plug-in API. If the goal here is to either generate or be a VST plug-in, doing VCVRack ones as well would be a big win. Also supporting LV2 in addition to VST would be good. (VCVRack info here: https://vcvrack.com/manual/PluginDevelopmentTutorial .)

mikeoliphant, 2020-11-09:
Dillon - would it make sense to put some parameters on circuit Inputs for this - like you did with V0dBFS? Maybe an input type (Audio/CV/Gate)? I could make the VST check for that when it is enumerating inputs, and have it do MIDI->CV/Gate when those inputs are present.

dsharlet, 2020-11-10:
Is the thing needed here several Inputs that are tied together conceptually? 

If so, one possibility is to just use existing Input components, with names that mean something. But this is pretty sketchy, another possibility is adding a component type with several terminals:
* CV
* Gate
* (Maybe?) Trigger
* Ground

Do circuits need more than one of these? If not, the special names approach is a bit more reasonable: you could just have an Input with the name CV, an Input with the name Gate, etc.

I think the best thing to do here would be to find some examples of circuits we want to simulate and see what is needed to do it cleanly. The one @mikeoliphant linked earlier (https://www.allaboutcircuits.com/projects/diy-synth-series-vco/) looks doable with ideal op-amps, *maybe* with non-ideal opamps. Many of the circuits I've been finding online look a lot bigger though and might be tough to simulate in real time...