dsharlet/LiveSPICE #95: Is .Net core faster?

dsharlet, 2020-11-04:
@Federerer mentions that switching to .net core made things faster. I'm out of the loop on .net framework developments since ~2013...

Federerer, 2020-11-07:
Do you have plans to modify .csproj files in the next few days? Because i wanted to migrate them to the new format, and I'm trying to avoid merge conflicts :)

dsharlet, 2020-11-07:
I'll avoid modifying the project files, thanks for offering to do that. I looked into it a little and did some prerequisites (migrated package.config to PackageReference), but hadn't gotten as far as the projects themselves.

dsharlet, 2020-11-17:
The answer to this is: yes, .net core 3.1 (and .net 5.0) are much, *much* faster. Both circuit solving and simulation are up to 2x faster vs. .net 4.8!

I'm going to close this, and start a separate issue for switching to .net 5.0

closed 2020-11-17 by dsharlet