dsharlet/LiveSPICE #96: Simulation delegate overhead?

dsharlet, 2020-11-04:
@Federerer mentions he found the simulation delegate to have significant overhead.

dsharlet, 2020-11-07:
I calculated the overhead by varying the block size of the benchmark on a medium size circuit:

Samples/Block | Rate (kHz)
---------------- | ----
2^6 | 210
2^10 | 215
2^12 | 215

Seems like the overhead is 2-3% when using 64 samples/block (1.4ms at 44.1 kHz). 64 samples/block is pretty small (though I know people do use this setting) and 2-3% overhead isn't bad, so I think this overhead is negligible.

Federerer, 2020-11-07:
I haven't said that the overhead is significant, just that there is some and it's related to the usage of DynamicInvoke() ;) Also, creating array for parameters adds another memory allocation to the audio signal path. It looks like it's unnecessary in this case. I've created a pull request with my changes.

dsharlet, 2020-11-07:
Oh, I see now, what you are proposing is also better for type safety anyways even if the overhead isn't huge. Thanks for the PR, I'll take a look now.

closed 2022-10-28 by dsharlet