dsharlet/LiveSPICE #98: Support speaker cabinet filters?

dsharlet, 2020-11-04:
@Federerer mentions he has been working on building a speaker cabinet impulse response into the simulation. This is interesting, but especially now that we have the VST plugin, my feeling is it would be best to keep the built-in GUI dedicated to just circuit simulation/debugging, and let this kind of simulation fall to other effect plugins...? @mikeoliphant also might have thoughts on this.

mikeoliphant, 2020-11-04:
Yeah, I think this is much less important now that we have the VST, which makes it easy to use a cabinet IR plugin.

Federerer, 2020-11-08:
That's right, except that you can't use them when you are editing in the standalone application. There are some possibilities here:
- Adding this functionality to LiveSpice directly,
- Adding ability to use vst plugins inside the LiveSpice making it a vst host.
- Adding the posibility to edit a schematic directly in vst plugin.

I think that the third option might be the best solution. This is also mentioned in #103 but i would implement this slightly different, e.g. when you open the schematic, you can choose between two modes: editing, with sound processing bypassed and processing, with editing disabled. Something similar to the current behavior of the standalone application, but in single window.