dsharlet/LiveSPICE #99: Use SIMD to optimize some things

dsharlet, 2020-11-04:
Row reduction on numeric types (as opposed to symbolic `Expression`s) is fairly SIMD friendly. However, when I've profiled this, `RowReduce` doesn't seem to take a lot of time. The other big row reduction happens at circuit build time on symbolic expressions, so I'm not sure SIMD helps. However, there might be other opportunities for vectorization as well after possible changes mentioned in #94.

Federerer, 2020-11-08:
I've tested that, and for the current matrices sizes difference is negligible.
I've also did some tests with the VisitSum method in the CompileExpression.cs replacing some sums of products with the Vector.Dot operation and it basically works, but to get any performance improvement all the terms should be stored in array form and not in separate variables, because array creation overhead wastes all the performance gain given by the vectorization.

dsharlet, 2020-11-08:
That makes sense. I think #97 might be a prerequisite for this. What I want to do with that issue is make a lot of the math expressed as matrix ops instead of big scalar expressions where possible. That should cause many of the things that are currently stored in many scalars stored in vectors/matrices instead. It should also make the generated programs a lot more compact and easier to optimize (both for the compiler and for humans). It should also make a lot of computer algebra operations faster hopefully.