dsharlet/LiveSPICE #112: Hardware Recommendation

TacoSlease, 2021-01-05:


TacoSlease, 2021-01-05:
Looking for some feedback on what some user are using for ASIO hardware. Any pros or cons that I should be aware of?

dsharlet, 2021-01-05:
I was using a Focusrite Scarlett 2i2 for a while and was pretty happy with it. It didn't give me any trouble.

CFH1313, 2021-01-05:
I'm using an RME fireface UC (not UCX) and I've had no issues at all. It's actually pretty awesome to be able to build stuff and test it with the VST. I've actually really enjoyed being able to do that. I've had some issues with graphical resizing where certain windows seem to hang, but removing the plugin and re adding fixes it. I found in general removing the plugin and starting fresh when changing circuits/patches seems to make it run much smoother with less issues.

mdouchement, 2021-01-05:
I'm using a Komplete Audio 6 mk2 and it works well. It comes with dedicated ASIO drivers provided by Native Instruments.

mikeoliphant, 2021-01-05:
What are you wanting to do with it? If you want to plugin in a guitar, make sure the interface has a high impedance ("instrument") input. Most do. If you want to play and monitor the output live, you want a device with solid ASIO drivers that can run at low buffer sizes for minimal latency (although this also depends a lot on your computer and the complexity of the circuit you want to simulate). Again, most devices should be fine.

TacoSlease, 2021-01-05:
Looks like I am going to give the Focusrite Scarlett Solo a try. It had many many good reviews on Amazon.

I am just going to design some FX and see how they function before I throw parts at a breadboard.

Thanks for your input, crew.

TacoSlease, 2021-02-09:
So I purchased a Focusrite Scarlett Solo and the audio configuration tests work, but every circuit I open is nothing but feedback.....even with no source.