dsharlet/LiveSPICE #133: Oversample set to 0 Produces Infinite Loop

RobertKeyser, 2021-09-19:
When Oversample is set to 0 by manually typing the value into the simulation window's input, an infinite loop occurs where it continuously tries to rebuild the simulation. There is no way (that I found) to get out of it besides killing the process.

Federerer, 2021-09-22:
Related issue: 'Iterations' are triggering simulation updates when you are typing a new value, which is annoying. Solution used in VST plugin (with two `ComboBoxes`) seems to work better. Side-note: from my personal experience, the default value of max 8 iterations is a little low, usually I usually use '32' on heavy distorting tube-based circuits 😃

dsharlet, 2021-09-28:
Thanks for reporting the issue(s). I've added some simple logic to just fix invalid values. I agree the typing triggering updates is annoying, but I do want to keep the ability to type in values, I've sometimes experimented with large values at least. I'll look and see if there's a way to fix updates on each typed character.

dsharlet, 2021-09-28:
It looks like I already tried to make the behavior only to update after changing focus (`UpdateSourceTrigger=LostFocus`), but that's not working for some reason...

Federerer, 2021-09-28:
`LostFocus` needs losing focus 😉 You need to click-away somewhere, which is not always the best solution. Maybe `Delay` would be better choice here? It counts time from the last keystroke, so you can type in a longer value without triggering simulation update. Mergeable example here #141