dsharlet/LiveSPICE #161: Simulation Diverged Error

ilenker, 2022-10-25:
First off this piece of software is simply amazing, thanks for creating it!

I'm just a layman trying to play around with circuits and trying to copy certain pedals digitally.

On some schematics I'm getting this error:

Error running circuit simulation.
Simulation diverged near 1 = 0 s + 0

This happens in the VST and the standalone program, with various oversampling and iteration settings.
If it happens once in the VST, it seems to persist with any schematic I load, until I make a new instance of the VST.

Some schematics do this from the start, and others do it with certain settings.

dsharlet, 2022-10-25:
It's hard to say for sure without seeing a particular schematic. Do you see the same problem in the example circuits?

There are a few basic reasons why this can happen:
- Really "stiff" circuits, such as high distortion circuits, can be numerically hard to simulate. In these cases, increasing oversampling and iterations can help.
- There's a short circuit or other problem in the schematic. A circuit that would fail in reality due to a short circuit will probably fail with "simulation diverged".
- The circuit may have initial conditions that are hard to solve for, and then running the dynamic simulation is too far away from the typical operating conditions of the circuit, and the first iteration fails to converge.
- Maybe just a bug somewhere.

If you have a circuit that reliably reproduces the problem that you believe should simulate OK, I'd be interested in taking a look when I get a chance, especially if it doesn't appear to be a particularly stiff circuit.

> If it happens once in the VST, it seems to persist with any schematic I load, until I make a new instance of the VST.

This sounds like there is a bug of some sort in the VST plugin where bad states persists across different circuits.

ilenker, 2022-10-25:
It can happen with the example circuits, but they are much more stable. The only way I can make it happen with the examples is maxing out the OS and the iterations (I use reasonable settings otherwise so it's not a problem with the examples).

The circuits I tried that have this problem are many of the ones found here: 
https://github.com/Federerer/livespice-circuits/tree/main/Orange

[Marshall JCM800 2203 preamp modded.txt](https://github.com/dsharlet/LiveSPICE/files/9863366/Marshall.JCM800.2203.preamp.modded.txt)
[Orange Rockerverb 50 Preamp.txt](https://github.com/dsharlet/LiveSPICE/files/9863367/Orange.Rockerverb.50.Preamp.txt)

(I just changed these from .schx to .txt files to be able to upload them here)

I'll try some more and see if the other circuits do this as well.

Federerer, 2022-10-25:
> If it happens once in the VST, it seems to persist with any schematic I load, until I make a new instance of the VST.

It's 100% a bug. I noticed that too and even fixed it locally on one of my branches.

And replying to the original question: In some cases reordering components might help, as doing so, usually creates different solution, which might be more stable. I just created a quick and dirty solution for it, that randomly creates a new permutation of components when that exception happens, and then I'm updating the file when closing the simulation window and there was no exception. All the schematics in my repository should work without any tweaking, but... on my branch, which uses tweaked tube models, so it might require reordering on master. 😉

Federerer, 2022-10-26:
> Marshall JCM800 2203 preamp modded.txt
this one contains switches, and there is a bug that causes initial state calculation to fail when there are unconnected nodes:
#148

dsharlet, 2022-10-27:
It makes sense that a circuit could fail to simulate if the initial conditions can't be solved for. Let me take a look at #148's reproducer. @Federerer do you think the ordering matters even when the initial conditions are successfully solved for?

dsharlet, 2022-10-27:
#162 may be the fix for the simulation diverging, if the issue is indeed switches causing the initial conditions to fail.

Federerer, 2023-04-04:
>@Federerer do you think the ordering matters even when the initial conditions are successfully solved for?
Yes, It's because when we have more than one pivot with the same value, we are just taking the first one, so we are getting different solutions for different orders of equations. I've also tried to use initial conditions and parameter values (I'm testing it with dynamic parameters enabled) as pivot conditions, but something is wrong with the `Evaluate` function, as it ends up in an infinite loop, trying to substitute those values. This needs more investigation 😉

dsharlet, 2023-06-14:
Which circuit are you running? This error can have many different causes. While we fixed a few particular issues in the latest release, there are definitely going to be other circuits with other issues that need looking at.

Your particular case is interesting because it's unusual to have the simulation diverge more than a split-second into the simulation. If you can share your circuit, I can take a closer look.

jhsa, 2023-06-14:
> 

Yes, no problem. Here is the circuit. Please remove the ".txt. at the end of the file. And thank you for looking into this.

[MyEQ_Baxandall.schx.txt](https://github.com/dsharlet/LiveSPICE/files/11750597/MyEQ_Baxandall.schx.txt)

jhsa, 2023-06-14:
Ok, something that might help. If I keep changing one pot only it seems to work. But as soon as I touch another pot, the error is displayed just before building the simulation.

jhsa, 2023-06-14:
Ok, even moving only one pot I can make it happen. 
Also, I have just started the simulation, left it for a couple minutes, then came back, then Moved one of the pots a couple times, and the following error was displayed just before it built the simulation again.

`Error: Simulation diverged near t = 230 s + 0`

Please note the "230s"

Federerer, 2023-06-14:
I think it's because after each pot change the solution is rebuilt, but simulation time is not set back to zero. My wild guess is that this might be related to the fact that some pot settings might produce big rationals like in https://github.com/dsharlet/LiveSPICE/issues/177

jhsa, 2023-06-14:
I have been playing with the examples. No errors as far as I could tell.  So it must be something triggered by my circuit.
I have just tried removing the first part of my circuit and simulate only the Baxandall EQ. It works much better, but it also displays the error after a while. I have noticed that when there is no error, the simulation build popup window doesn't appear when moving the pot. 
But when the error show up, this simulation build popup window appears on the screen.
I have to bring the oversample down to 2 for the simulation audio to work ok.

dsharlet, 2023-06-15:
I've been playing with this circuit and haven't been able to get it to diverge yet. I did find something that might at least cause the circuit to not behave as intended: capacitors C2, C4, C5, and C6 are all stacked right on top of each other. This is the first capacitor leading from the input to the first op-amp.

This probably means you actually have 4 capacitors in parallel, which did actually have some issues I recently fixed in #182. That said, that fix is in the latest release, so this shouldn't actually be causing the problem.

I do also see that this circuit is fairly computationally intensive especially when a transient is in the input signal, so needing to reduce the oversample factor isn't surprising.

It is *possible* that an older (slower) CPU could exacerbate issues like this, because if it drops samples due to failing to keep up with the processing demands, the dropped samples introduce large transients. Which can then cause the processing to get even slower, leading to more dropped samples, and if the circuit is "stiff", this can lead to a failure to converge. However, the circuit doesn't sound that stiff to me, so I'm not sure this is relevant at all.

dsharlet, 2023-06-15:
It is a bit surprising this circuit is as slow to simulate as it is. It doesn't have that many non-linear components, and I thought op-amps were one of the cheaper components to simulate.

One option for you is that I don't think you are anywhere near saturating the op-amps, which means you can probably use an ideal op-amp in simulation without any audible differences. I think that would actually make this circuit entirely linear, which should make it simulate *dramatically* faster.

jhsa, 2023-06-15:
> I've been playing with this circuit and haven't been able to get it to diverge yet. I did find something that might at least cause the circuit to not behave as intended: capacitors C2, C4, C5, and [...]

The problem with the stacked components was due to trying to copy them. When I hit paste, it copied them on top of the original and not where I clicked. perhaps this could be changed as it  seems to be a bit more intuitive and it would avoid this kind of situation?
I did discover this problem and fixed it. But I still have the diverge error problem. By removing the first part of the circuit and simulating only the EQ part of the circuit, it runs much better, but I still had the error happening a couple times.
On this machine I do need to reduce the Oversample value to 2 or even 1.  The example "Phase 90" doesn't simulate with audio on my PC, even with the Oversample at 1.

jhsa, 2023-06-15:
Ok, I also see the error when seeing rthe "Generic Opamp". Next will change the circuit to use  the ideal opamp. But i would like to know how to set the power supply up for this. I am new to Live Spice, and I do apologize.

jhsa, 2023-06-15:
Ok, some good news. I don't know if I did it right but I used the Ideal Opamp and biased it with 12V (Supply should be 24V). The simulation worked without even the need to decrease the Oversample number. It worked with it set to 8. No errors reported.
So does it mean that the Opamps do take a bit more computing power than expected? ;) 
Here is the file attached. Please remove the ".txt" at the end of the file name.
Thank You..!

[MyEQ_Baxandall_Ideal_Opamp.schx.txt](https://github.com/dsharlet/LiveSPICE/files/11759772/MyEQ_Baxandall_Ideal_Opamp.schx.txt)

jhsa, 2023-06-15:
So, is there anything that can be done, or the problem is my PC that is too slow?
I might be able to try another machine with an i5 processor and 8Gb of RAM. It still has windows 7 on it though.

dsharlet, 2023-06-15:
I will take a closer look at this simulation soon, but I wouldn't expect a lot. I'd be surprised if there were a way to improve it that much.

BTW, I'm a bit confused about the change you made. It seems like when going from the non-ideal op-amps to the ideal op-amps, you could just remove the power supply and otherwise leave the circuit as-is? It seems to work fine when I try that.

jhsa, 2023-06-15:
I want the bias at half supply. I actually want to use a symmetrical power supply.
And I have bad news. changed some values, and now the simulation doesn't even start. I have the diverging error at 0s. I can't understand this. :(

jhsa, 2023-06-15:
I had to download the file I posted here. the one I had became corrupted somehow and it would throw the diverging error all the time.

Could you please try this file?  It throws another error when I flick the switch. It happens also if I leave one pin of the switch unconnected.

`Exception: Index was outside the bounds of the array.`

[MyEQ_Baxandall_Ideal_Opamp_Sswitch.schx.txt](https://github.com/dsharlet/LiveSPICE/files/11760779/MyEQ_Baxandall_Ideal_Opamp_Sswitch.schx.txt)

dsharlet, 2023-06-15:
This one is working fine for me, even after flipping the switch. Are there any other steps required to reproduce the issue?

Can you also share the corrupted file? I am curious as to exactly what happened there.

jhsa, 2023-06-16:
> This one is working fine for me, even after flipping the switch. Are there any other steps required to reproduce the issue?
> 
> Can you also share the corrupted file? I am curious as to exactly what happened there.

Try disconnected one of the SPDT switch pins, like using only two of them as with a normal switch. It won't let me this way.  It will produce the error:

`Exception: Index was outside the bounds of the array.`