dsharlet/LiveSPICE #164: Non-realtime mode

Macflash, 2022-11-05:
Is there anyway to use this in a non-realtime fashion? With larger circuits there is no way for me to get it to work in realtime, resulting in painful audio pops and terrible results. For designing a circuit usually I would play a similar lick on repeat, so it would be nice to be able to let the simulation take however long it needs and get a clean, listenable result, even if it takes a few seconds to process.

domdec314, 2022-11-06:
LTSpice will let you process a WAV file. I’d imagine it wouldn’t be simple to add this feature to LiveSPICE as it’s completely built around live audio use.

dsharlet, 2023-04-30:
There's recently been discussion of command line version of LiveSPICE (#171), which would do this.

The VST plugin probably also lets you do this, because DAWs have a lot more functionality than LiveSPICE. Offline rendering hasn't been a priority because you can just use LTspice or the like, but I get that those tools are a lot more complicated and can be a pain to use.