dsharlet/LiveSPICE #166: Sinusoidal voltage source parameter

Stratplayer57, 2022-12-18:
Hello,
I am pretty new to LiveSPICE (already accustomed to LT Spice and similar).
I was trying to model a circuit when I encountered a problem right at the start. By instantiating a voltage source,  the default parameter for voltage is: "Sin[628,318530717959*t] V" which I assume is in the form Sin[Amplitude, omegat*t] V". 
So I replaced it by the needed values and it appears that by pressing enter, the order of the parameters changed to:  "Sin[t*230,314] V".
When simulating, I get the following error: Exception: Unresolved name " 'Undefined variable '230,314'.'. ".

I am sure there is something trivial that I haven't seen. Could someone help me?

PS. If someone knows a method to import already existing LTSPice schematics, I would be very grateful.

Thanks in advance,
Théo

dsharlet, 2022-12-19:
I think you are encountering a bug due to EU vs. US style decimals. The default is supposed to be `Sin[100*2*Pi*t] V` = `Sin[628.318 t] V`, but something is going wrong and interpreting it incorrectly.

However, I suspect that what you actually want is an "Input", which is a voltage source where the voltage is proportional to an audio input. The general voltage source with a sine wave input is mainly only useful for testing or experiments. I'll have to look and see what is going on with such inputs if that's what you were actually trying to do.

It would help if you can attach the `.schx` file that is causing problems to the bug, so I can see exactly what the voltage field ended up being set to.

Federerer, 2023-05-08:
I've noticed that too. We should ensure that we are always saving files using `NumberFormatInfo.InvariantInfo` and accepting any number separator when parsing user input or file (this is a good practice for any kind of interface or API - be specific what you provide and be flexible what you accept).

standard-librarian, 2023-10-27:
Is there a way to know what are the valid expressions, i mean is there such a thing as Cos[628*t], Pulse[628*t] or Square[628*t] for example ?