dsharlet/LiveSPICE #170: Issue with N-Channel JFETs

rossbalch, 2023-03-02:
Exception: Incorrect number of arguments supplied for call to method 'Double lambda_method(System.Runtime.CompilerServices.Closure, Double, Double)'

rossbalch, 2023-04-19:
[https://drive.google.com/file/d/1TlxaD21_74qfYGA1rd_gQmbTEyMsgP9w/view?usp=sharing](url)

Link to schematic that results in this error

Federerer, 2023-04-19:
Hello, I did some testing and the good news is that there is no exception on my current branch. The bad news is that the JFET model is very unstable. I simplified your example to just single stage and only this way I could get it to converge, but only for small signals. As soon as I try to make it clip id diverges, so my guess is that there is something wrong with the model itself. I've never tested JFETs before, only BJTs and tubes, so it requires some investigation 🙄 BTW, that's quite a big circuit 😁

rossbalch, 2023-04-21:
Yeah seems the Jfet circuits are a toughie. This one is the lead channel of a Mark IV converted to Jfet instead of tube so it is a more complex one.

dsharlet, 2023-05-01:
I've started looking at this issue, sorry for the delay. 

I think the thing that is causing problems here is C25. It's a capacitor at the upper right part of the circuit that is connected directly from 9V to ground. While this seems like it shouldn't break anything, it is weird and won't contribute anything to the simulation. If I remove that component, the simulation at least solves, but it immediately diverges. That's going to be a trickier issue to debug.

When I was looking at this circuit to simplify it for debugging, I noticed that there are many large chunks of the circuit that are just combinations of linear components. There *shouldn't* be anything wrong with this, but it will make the circuit bigger and more complicated to simulate or at least solve. However, like the capacitor that is connected between two DC voltages, I wonder if there are some other suspicious/problematic components that aren't as easy to spot?

Regardless, the error that LiveSPICE gives in this case is terrible, and I will try to improve that. The thing that's happening is that the solver doesn't fully solve the circuit, so when it goes to simulate it, it tries to simulate with functions that can't actually be executed.

dsharlet, 2023-05-08:
I believe this circuit has a few issues. One of them is a bug, fixed in the above commit.

The other issue is the capacitor and diode directly connected from 9V to ground. After removing those two components (which shouldn't do anything anyways), and the above commit, the circuit simulates without diverging.

dsharlet, 2023-05-08:
Going to keep this open because I think this circuit should still simulate OK with the diode and capacitor in place. I was thinking that the diode would have absurd current going through it, but it's the other way around so it should be fine. And the capacitor should be fine either way.

rossbalch, 2023-05-08:
Yeah, thanks for looking into this. This is a verbatim circuit, so it probably does contain a few unnecessary power filtering elements.

rossbalch, 2023-05-12:
Another thought. Those JFETS are all connecter to pots for bias adjustment. In real life tollerances mean the bias voltage needs to be set for each J201. However, the necessary voltage will be precise here because they're all the same model (I'm too dumb to figure out what that is though). Could the simulation be improved by replacing the pots with a resistor of a specific value?

fcoberrios, 2023-06-25:
Can i ask, so right now this is working? Can i use a few N-Jfet stages with a pot for every stage for controlling the bias during the simulation on the vst without crashing the plugin or something? :)
Thanks!

dsharlet, 2023-06-25:
I don’t think any of the issues being looked at in this thread are actually related specifically to JFETs, they should be working.

commit 3adf1a3f (2023-05-08): Pull in huge rational fix to fix #170 and #177