dsharlet/LiveSPICE #172: Current release is very out of date

mikeoliphant, 2023-04-18:
A lot of changes have been made since the last "release" version (which is from 2020!). Maybe it would make sense to cut a new version for people who aren't building from source?

Federerer, 2023-04-18:
That's definitely the way to go. I was thinking of using a github action too. @mdouchement maybe create a PR if you already have a working solution?

mdouchement, 2023-04-18:
The GH's workflow needs some adaptations, if everyone is OK, I can do it.

mikeoliphant, 2023-04-18:
I'm all for a github workflow solution.

It would allow for other useful things as well - like test build on push/PR.

mikeoliphant, 2023-04-18:
> Do we want snaphots on merge on master or release on new release created?

I think the best thing is to have separate actions for test builds and snapshot (release?) builds. Test builds get triggered by push/PR on master, with releases being manually triggered.

mikeoliphant, 2023-04-18:
> Usually you trigger "test" pipeline on PR created/updated, a "dev" release pipeline on every PR merged to master and a normal release build when a version tag is created on master.

Depends on the repo, and whether the main branch is protected. The vast majority of times I've submitted PRs it has been against main/master. I think that would be fine here, too.

dsharlet, 2023-04-18:
@mdouchement it would be amazing if you wanted to provide some GH actions. I think just any one of the proposed ones would be fine to start with (tests, building a new release, etc.).

I took a quick look at the actions you built on your fork. It looks reasonable to me, but correct me if I'm wrong, this doesn't run inno setup to build an installer? Not sure if it would be possible to do that (we'd have to get inno setup installed in the GH action environment somehow). Alternatively, the inno setup configuration is very, very simple, we don't need to use it. Maybe there's a better way to make installers now (10 years later...).

We definitely need to do a newer release, as the issue says!

mikeoliphant, 2023-04-18:
> took a quick look at the actions you built on your fork. It looks reasonable to me, but correct me if I'm wrong, this doesn't run inno setup to build an installer? Not sure if it would be possible to do that (we'd have to get inno setup installed in the GH action environment somehow).

I'm pretty sure inno setup is installed in the github windows runner:

https://github.com/actions/runner-images/issues/5006

Federerer, 2023-04-18:
> I'm pretty sure inno setup is installed in the github windows runner:

I've definitely seen it used by some repositories on github, so it has to be.

mdouchement, 2023-04-19:
> @mdouchement it would be amazing if you wanted to provide some GH actions. I think just any one of the proposed ones would be fine to start with (tests, building a new release, etc.).
>
> I took a [...]

My workflow only builds an archive with the compiled https://github.com/dsharlet/LiveSPICE/tree/master/LiveSPICE folder.

I can do these workflows:
- Check if a PR is buildable
- On merging on `master`, create a snapshot (ZIP archive) as an artifact of the workflow
  - e.g. https://github.com/mdouchement/community-edition/actions/runs/4742608205 (a GH account is needed to download artifacts)
- On a new release (e.g. https://github.com/mdouchement/community-edition/releases/tag/v0.8.1-fix-vulns)
  - Create an executable installer
  - Create a zip archive for "portable" installation

erikgitarristen, 2023-04-20:
Hello! Is it be possible to update the vst to read the files from the latest liveSPICE? It seems that the vst can't cope with the new electrical components in the latest liveSPICE, or am i doing something wrong? I've managed to use mdouchements workflow to build the latest standalone version.

dsharlet, 2023-04-30:
I'm putting together and testing a new release now. Sorry for the delay, it took me a while to realize the fix for `CompareTo` blocking #176. I thought it was worth fixing that before doing a new release.

I'm now working on the release notes. The release has a lot of contributions from several of you, I want to make sure I don't miss anything. Here's what I've got, please let me know if you think I missed anything notable.

The draft I have currently is:

* VST plugin upgraded to VST3, and some improvements and bug fixes to the VST plugin (contributed by @mikeoliphant)
* New vacuum tube diodes and pentode models (contributed by @Federerer)
* New diode, transistor, and triode components (contributed by @mdouchement)
* New circuit examples (contributed by @Federerer)
* New potentiometer sweep types (contributed by @mdouchement)
* Significant performance improvements due to upgrading to .net6.0 and optimizations (contributed by @Federerer)
* Various bug fixes and improvements

The automated release notes from the last release are here, if you want to reference them:
```
## What's Changed
* Fix initial directory of open file dialog for loaded plugin instances by @dsharlet in https://github.com/dsharlet/LiveSPICE/pull/101
* Row reduction row swapping by @Federerer in https://github.com/dsharlet/LiveSPICE/pull/107
* Marshall examples by @Federerer in https://github.com/dsharlet/LiveSPICE/pull/108
* Migration to the new .csproj format by @Federerer in https://github.com/dsharlet/LiveSPICE/pull/105
* Strongly typed process delegate by @Federerer in https://github.com/dsharlet/LiveSPICE/pull/104
* Vst3 branch integration by @mikeoliphant in https://github.com/dsharlet/LiveSPICE/pull/114
* Multithrow switch support for VST by @mikeoliphant in https://github.com/dsharlet/LiveSPICE/pull/120
* Display marking letter on the potentiometers by @mdouchement in https://github.com/dsharlet/LiveSPICE/pull/118
* Ground symbol as a named wire by @Federerer in https://github.com/dsharlet/LiveSPICE/pull/128
* ComputerAlgebra as submodule by @Federerer in https://github.com/dsharlet/LiveSPICE/pull/131
* Schematic rendering misalignment by @Federerer in https://github.com/dsharlet/LiveSPICE/pull/138
* Potentiometer grouping by @Federerer in https://github.com/dsharlet/LiveSPICE/pull/134
* Add 6SL7GT triode model by @mdouchement in https://github.com/dsharlet/LiveSPICE/pull/142
* Binding delays in simulation window. by @Federerer in https://github.com/dsharlet/LiveSPICE/pull/141
* Diode and Pentode model by @Federerer in https://github.com/dsharlet/LiveSPICE/pull/121
* Fix for #149 by @Federerer in https://github.com/dsharlet/LiveSPICE/pull/150
* Fix for PotControl limits. by @Federerer in https://github.com/dsharlet/LiveSPICE/pull/152
* Add several Spice models by @mdouchement in https://github.com/dsharlet/LiveSPICE/pull/155
* Add AntiLogarithmic & Sigmoid potentiometer taper by @mdouchement in https://github.com/dsharlet/LiveSPICE/pull/156
* Update VST to use AudioPlugSharp NuGet packages by @mikeoliphant in https://github.com/dsharlet/LiveSPICE/pull/167
* Command line parameters parsing in Test, Gaussian elimination benchmarks. by @Federerer in https://github.com/dsharlet/LiveSPICE/pull/124
* Github actions for builds and releases by @mdouchement in https://github.com/dsharlet/LiveSPICE/pull/174
* Fix switch initial conditions by @dsharlet in https://github.com/dsharlet/LiveSPICE/pull/162
* Enable running release manually by @dsharlet in https://github.com/dsharlet/LiveSPICE/pull/175
* Update SDK to net6.0-windows (fixes #176) by @dsharlet in https://github.com/dsharlet/LiveSPICE/pull/179

## New Contributors
* @mdouchement made their first contribution in https://github.com/dsharlet/LiveSPICE/pull/118

**Full Changelog**: https://github.com/dsharlet/LiveSPICE/compare/v0.13...v0.14
```

mikeoliphant, 2023-04-30:
> * VST plugin upgraded to VST3, and some improvements and bug fixes to the VST plugin (contributed by @mikeoliphant)

VST was already VST3, so I'd just say "improvements and bug fixes to the VST plugin".

mikeoliphant, 2023-04-30:
> > * VST plugin upgraded to VST3, and some improvements and bug fixes to the VST plugin (contributed by @mikeoliphant)
> 
> VST was already VST3, so I'd just say "improvements and bug fixes to the VST plugin".

Hmmm - I just looked and you were correct. VST3 *is* new in this release. It really has been a long time since the last release...

dsharlet, 2023-05-04:
I just finally did the new release :) Here it is: https://github.com/dsharlet/LiveSPICE/releases/tag/v0.14

It would be great if you guys could try it out and report any issues you see. I did some testing of it of course, but it would be a good to hear it is working well for others.

mikeoliphant, 2023-05-04:
Great!

mikeoliphant, 2023-05-04:
I did some cursory testing - mostly on the VST - and it looks good to me.

closed 2023-05-04 by mikeoliphant