dsharlet/LiveSPICE #186: Unable to install in Windows-for-ARM inside Parallels

ggkountouras, 2023-05-05:
**Platform**

Windows 11, Parallels 18.2.0, macOS 13.1

**Error**

I'm getting the following error with [Preview 14](https://github.com/dsharlet/LiveSPICE/releases/tag/v0.14), even though Windows-for-ARM should be able to emulate x64 apps. I don't expect real-time performance from emulation inside a VM, since that is 2 layers of indirection.

[image]

**Possible fixes**

Newer versions of .NET (.NET Core, not the older .NET Framework) support cross-platform apps, including ARM architectures natively. That could be an avenue to cover Windows-for-ARM as well.

Federerer, 2023-05-05:
It's definitely doable by adding another build target. However, I don't have any hardware to test it. Are you able to compile it yourself to check if it works?

Federerer, 2023-05-05:
Probably just like you described. you just have to add a new configuration (ARM64) in configuration manager: 
[image]
(simply select x64 as a base, you don't have to manually edit anything).