dsharlet/LiveSPICE #187: Schematic cannot be displayed in multiple editors simultaneously.

Federerer, 2023-05-06:
Currently we are using `Tag` property on schematic controls to link between component view <-> model, which causes issues when you try to load the same schematic in multiple editors. To avoid this we are making clones:
```cs
var clone = Circuit.Schematic.Deserialize(editor.Schematic.Serialize(), Log); 
```
which is very slow (I'm getting a 5s ui lag moderately big schematic). So we should probably switch to something different e.g. keeping a `Dictionary` in `SchematicControl`

dsharlet, 2023-05-06:
I just looked at `Deserialize`, because I've been annoyed how slow that is too sometimes. I think it has a ridiculous performance bug that should be easy to fix. We rebuild all the nodes every time each element of the circuit is added to the schematic...

dsharlet, 2023-05-06:
Also, #128 contributed to this in a weird way, because we do a lot of circuit graph algorithms when named wires are changed.

I was being really lazy on this code the first time around because I didn't think it would matter. I'm working now on rewriting this to use proper graph algorithms, which should make them way, way faster.

dsharlet, 2023-05-06:
Please take a look at #189 and see if you still think we need to avoid the cloning. I agree the cloning is not ideal, but it's simple, and maybe it's not a problem if deserialization isn't absurdly slow.

closed 2023-05-07 by dsharlet