dsharlet/LiveSPICE #191: Add ideal VCA component

tommiot, 2023-05-08:
An ideal VCA component with a gain/voltage parameter would be useful for designing compressors, gates etc. These are typically current in / current out devices where the current relation is Iout/Iin =10^(Vc*K) where Vc is the voltage at control port. 

See for example SSI2164 or THAT2180 datasheet for reference

dsharlet, 2023-05-09:
I think adding a component for this makes sense. I'd need to do a bit more reading of the datasheets to learn exactly what the model would be.

BTW, if you are itching to have one of these right away, we do have everything you would need to implement this as a schematic: the "Named Voltage" and "Named Current" components can be used with voltage/current sources to produce voltages/currents with arbitrary expressions. The Op-amp model example shows how this is done in that case, I think this component would be similar.

[image]

tommiot, 2023-05-09:
Thank you @dsharlet, I wasn't aware there were abitrary voltage and current sources capable of this. They works for me for simulating a particular circuit I was searching the VCA model for. 

I never bothered modeling the VCA internals when designing compressors or noise gates or voltage controlled filters with VCAs in SPICE. They have relatively low noise and distortion anyway. I think an ideal VCA model would be enough for most people.

Here's the VCA model I've used:

```
.SUBCKT VCA_SSI2164 in out EC- VGND
V1 in VGND 0
* Models 10k impedance for the VC input
R2 EC- VGND 10k
* Stability network
R3 in 1 560
C1 1 VGND 560p
* Current gain expression
G1 OUT VGND VALUE={-I(V1)*PWR(10, -V(EC-) * 1.5)} 
.ENDS
```

Federerer, 2023-05-09:
> BTW, if you are itching to have one of these right away, we do have everything you would need to implement this as a schematic: the "Named Voltage" and "Named Current" components can be used with voltage/current sources to produce voltages/currents with arbitrary expressions.

That's good to know. I've always wanted to try some compressors and VCF's 😁
And it works as intended:
[image]

Federerer, 2023-05-09:
This stuff is great, I already found two bugs :) One in computer algebra - this circuit does not work without the delay buffer:
[VCA.zip](https://github.com/dsharlet/LiveSPICE/files/11436945/VCA.zip)
And the other one in my optimization - adding a delay buffer breaks my dependencies searcher, because it does not recognize that if something depends on v_x[t-1] it also depends on v_x[t] 😉 Adding a probe to force it as output variable helps.
[image]

dsharlet, 2023-05-10:
I just pushed a fix for the computer algebra breakage, we were missing a rule for derivatives of Pow.