dsharlet/LiveSPICE #193: "Sequence has no elements" error

ludwigparker, 2023-05-13:
I built a simulation of a tweaked Marshall Super Lead preamp, and the simulation worked perfectly fine before. After the new update I decided to try and add a power amp to the end of it, using EL84s. I used the Minimalist 18-watt schematic, following it very closely (the only difference is that I simplified it by adding rails for the voltages A and C, as well as the voltage going into the EL84s). As soon as I load it, the message I get is "Sequence has no elements". I have never had this issue before. I don't think I'd have done anything wrong, but I will have to check my project against the original schematic when I get a chance
[image]

dsharlet, 2023-05-13:
Thanks for the bug report, sorry to hear about the breakage. Can you share the actual schematic that produces the error?

ludwigparker, 2023-05-13:
> Thanks for the bug report, sorry to hear about the breakage. Can you share the actual schematic that produces the error?

Here's a few screenshots. I've included the preamp schematic by itself, showing that none of it changed and still runs fine. In the VST version I get the same issue in a popup
[image]
[image]
[image]
[image]

dsharlet, 2023-05-13:
I was hoping you could share the actual file :) Reproducing such a large circuit exactly would take quite some time...

ludwigparker, 2023-05-13:
I apologise, I misinterpreted what you meant! How should I share it on here? It won't let me add it into this comment. Please forgive me, I'm new to Github honestly

dsharlet, 2023-05-13:
No problem. You can rename the file to have the extension `.txt` instead of `.schx`, and then you can attach it to a GitHub comment. Also, you can just open it in something like Notepad, copy all the text, and paste it in a code block (the `< >` icon in the github text editor).

ludwigparker, 2023-05-13:
Thank you for your wisdom. Here's the schematic (hopefully posted right)

`<?xml version="1.0" encoding="utf-8"?>
<Schematic Name="  " PartNumber="">
  <Element Type="Circuit.Symbol, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" Rotation="-5" Flip="false" Position="-730,-190">
    <Component _Type="Circuit.Input, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" V0dBFS="1 V" Name="IN" Description="Ideal voltage source representing an input port." />
  </Element>
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="-760,-190" B="-750,-190" />
  <Element Type="Circuit.Symbol, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" Rotation="-1" Flip="false" Position="-760,-190">
    <Component _Type="Circuit.Ground, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" WireName="GND" Name="GND1" Description="Nodes with the same name are connected as if they were connected by a continuous wire." />
  </Element>
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="-690,-190" B="-690,-170" />
  <Element Type="Circuit.Symbol, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" Rotation="0" Flip="false" Position="-690,-150">
    <Component _Type="Circuit.Resistor, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" Resistance="1 MΩ" Name="R1" Description="Standard resistor." />
  </Element>
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="-690,-130" B="-690,-120" />
  <Element Type="Circuit.Symbol, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" Rotation="0" Flip="false" Position="-690,-120">
    <Component _Type="Circuit.Ground, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" WireName="GND" Name="GND2" Description="Nodes with the same name are connected as if they were connected by a continuous wire." />
  </Element>
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="-710,-190" B="-690,-190" />
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="-690,-190" B="-670,-190" />
  <Element Type="Circuit.Symbol, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" Rotation="-1" Flip="false" Position="-650,-190">
    <Component _Type="Circuit.Resistor, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" Resistance="68 kΩ" Name="R2" Description="Standard resistor." />
  </Element>
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="-630,-190" B="-610,-190" />
  <Element Type="Circuit.Symbol, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" Rotation="0" Flip="false" Position="-560,80">
    <Component _Type="Circuit.Triode, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" Model="DempwolfZolzer" SimulateCapacitances="False" Mu="83.5" K="1.73E-06" Ex="1.4" Kg="1060" Kp="600" Kvb="300" Rgk="1 MΩ" Vg="330 mV" Gamma="1.26" G="0.002242" Gg="0.0006177" C="3.4" Cg="9.901" Xi="1.314" Ig0="80.25 nA" Cgp="2.4 pF" Cgk="2.3 pF" Cpk="900 fF" Name="V1A" PartNumber="12AX7" Description="" />
  </Element>
  <Element Type="Circuit.Symbol, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" Rotation="0" Flip="false" Position="-570,160">
    <Component _Type="Circuit.Resistor, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" Resistance="1.5 kΩ" Name="R3" Description="Standard resistor." />
  </Element>
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="-570,120" B="-510,120" />
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="-570,100" B="-570,120" />
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="-570,120" B="-570,140" />
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="-510,120" B="-510,140" />
  <Element Type="Circuit.Symbol, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" Rotation="-2" Flip="true" Position="-510,160">
    <Component _Type="Circuit.Capacitor, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" Capacitance="680 nF" Name="C2" Description="Standard capacitor component" />
  </Element>
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="-570,200" B="-510,200" />
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="-510,180" B="-510,200" />
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="-570,180" B="-570,200" />
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="-570,200" B="-570,220" />
  <Element Type="Circuit.Symbol, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" Rotation="0" Flip="false" Position="-570,220">
    <Component _Type="Circuit.Ground, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" WireName="GND" Name="GND3" Description="Nodes with the same name are connected as if they were connected by a continuous wire." />
  </Element>
  <Element Type="Circuit.Symbol, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" Rotation="-1" Flip="false" Position="-510,-10">
[... cut ...]
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="655,-170" B="655,-40" />
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="545,120" B="545,140" />
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="545,140" B="655,140" />
  <Element Type="Circuit.Wire, Circuit, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null" A="655,0" B="655,140" />
</Schematic>`

Federerer, 2023-05-13:
Adding a ground reference on the output side helps:
[image]
Also, set the speaker impedance to 8 ohm instead of infinity, and 0dBFS to around 25V to get a proper output level:
[image]
BTW: it's quite gainy circuit 😉 Also, I've noticed some voltage spikes around output transformer. From my experience, using a more complicated transformer/speaker model  instead of 'ideal' one usually helps. Check [this link](https://www.aikenamps.com/index.php/designing-a-reactive-speaker-load-emulator) for speaker reference. You can also google "transformer t model" or check my examples [here](https://github.com/Federerer/livespice-circuits).

Federerer, 2023-05-13:
@dsharlet after quick debugging: it looks like ComputerAlgebra produces some empty equations:
[image]
I'll take a more in-depth look later.

Federerer, 2023-05-13:
OK, as I thought, we can't really solve for voltages on the output side, as we are missing a ground reference. Using somehow more 'realistic' transformer (just adding some small capacitance between primary and secondary) makes this solvable without the need for secondary ground. We should just catch this error earlier and provide a better message for the end user instead of failing at the code generation stage with a cryptic error. 😉

ludwigparker, 2023-05-13:
Pardon the late reply, but earlier I did those tweaks that Federer suggested. It worked, the simulation ran but it wasn't very stable. I kept getting the "simulation diverged" error after just a few seconds of playing guitar through it. Not immediately, which leads me to believe that it may just be that my circuit/simulation is too unstable. 
I also have a version of this schematic where instead of being a higher-gain circuit with the first two stages in series (as is shown in my comments above), I run them in parallel. This allows me to have a much more stable simulation, and the only time it diverges or crashes is when I have the volume/gain at very high settings, and I'm playing the strings hard. Almost like overloading a real amplifier, but instead of catching fire my simulation just shuts off. This happens no matter what with the series version after a few seconds, so is it just high-gain circuits that make it crash?
Is there some kind of workaround for this? Some way to make it so that I can run a simulation without it "exploding"? 
I'm sorry if I come off as naive with any questions or responses, I'm honestly not very adept with coding and specific modeling design. I use this program to prototype amps and pedals, so while I may understand something on a circuit level, anything involving the actual software is beyond me but I am definitely open to learning it.

dsharlet, 2023-05-13:
It depends on why the simulation is diverging:
- If it's diverging just because the circuit is very "stiff" (high gain with hard-ish clipping), increasing the "Oversample" or "Iterations" options **might** help. The more oversampling there is, the more hard clipping looks like soft clipping :)
- If it's because there's a very large current/voltage in the simulation (a short circuit being an extreme example), more oversampling and iterations probably won't help. As you mentioned, if this is the case, the circuit would probably fry some components in real life :)

Agreed the error reporting is bad here, I've started working on improving this.

Federerer, 2023-05-13:
I was testing on my development branch, I just checked on release version and it's as you described. The triode models that you used are very stable, I've tested them with a square wave and lots of gain with no problems. Pentodes are using Koren's model which is less stable in such situations, at some point I've implemented some changes for those models which helps with stability, but it seems that this is still not merged to the master branch yet.

ludwigparker, 2023-05-13:
No worries Dillon, this stuff is such an enigma, it's frankly amazing that software like this is even possible. I probably sound like an old geezer with a computer saying that, but I'm just glad something like this exists in the first place, so I'll take it warts and all!
As for the settings, I tried a few different OS and iteration settings. My system can handle higher iteration settings, which tends to clean up the signal a bit, but with any oversampling stuff above 4x it gets pretty rough. My computer isn't the most up to date admittedly, but to even make this circuit run in the first place, I had to set my OS to 1 (with pedals, I can usually set to 4x and it works well, most amps are 2x, but 1x is all my computer can handle with a simulation like this). I'm led to believe that if I just had a better system, I could run higher quality simulations and be able to have them not crash. 
Federer, what do you recommend as far as settings for the pentodes? If applicable of course, I'm happy to wait for the newer release but if there's anything in the meantime I can try as a layperson, I'll be happy to be the guinea pig

Federerer, 2023-05-13:
> * If it's diverging just because the circuit is very "stiff"

Mostly that, plus not so smooth grid current equation (its either 0 or a resistive behavior for positive voltages) that creates a "jumpy" derivative at the transition point, which is usually bad and causes such problems. I've tried to fix that in https://github.com/dsharlet/LiveSPICE/pull/147 and it looks like I succeeded, because it works fine on my branch, but I'll retest that on that specific one from pull request cause maybe I've done some additional stuff later. 😉

> Federer, what do you recommend as far as settings for the pentodes?

You might try to reduce gain a little bit. EL84 doesn't need much voltage to be driven fully as opposed to 6L6 or EL34, currently you are blasting them with ~100V of signal from heavily overdriven PI.  I would've tried to use a PPIMV in this case.

Federerer, 2023-05-14:
Oh, @ludwigparker I forgot about the most obvious thing. Use higher iterations limit. In my case i had to set it to 12 to fully eliminate output popping. @dsharlet  I really hate this parameter, because setting it higher than needed does nothing, and setting it lower usually produces cracks and pops or just breaks the simulation. I've got much better results manipulating the relative/absolute error in newton iterations. 😉

Federerer, 2023-05-14:
It plays quite nicely 😁
https://youtu.be/IuQB2RMRTfQ
Although, it barely runs on my machine with reaper and obs studio processing and recording 😉

ludwigparker, 2023-05-14:
So glad you were able to record a demo with it! 
For the transformer/speaker modeling that you applied to the output section, what values do you recommend as a neutral starting point? I'm currently designing a few different physical amplifiers for friends/family/et cetera, so being able to transpose the baseline values across various power amp schematics would help greatly for getting the simulations more realistic. I wish I could just use my intuition, but I'm still new getting into the world of tube amp design, so any help is appreciated!

dsharlet, 2023-05-14:
I made a few fixes that catch this problem earlier. The circuit now fails because the simulation diverges, which is about expected.

I actually tried to catch the issue earlier (the last commit I tagged this issue with), but interestingly, there are example circuits that successfully simulate that seem to be underdetermined systems of equations! I had to revert that last check.

dsharlet, 2023-05-14:
I think between @Federerer's model fix and the better error checking, this issue should be fixed.

closed 2023-05-14 by dsharlet