dsharlet/LiveSPICE #198: Stops Loading Models, or creating new ones

jhsa, 2023-06-15:
I have installed the current latest version of LiveSpice. 
Very often, it stops loading models (examples included) or even creating new schematics.
I have googled the problem and found out that deleting the "User.config" file would solve the problem. 
Well, it does for some time, but then after using the app a couple times it happens again.
Thanks

Morophla, 2024-03-23:
I have the same issue. Have you found a solution?

Thanks!

jhsa, 2024-03-24:
Nope.. As far as I know we have to delete the  "User.config" file every time it happens.

Morophla, 2024-03-24:
Oh... I don't even find a User.config file! Maybe I don't search at the right place... Where is yours?

jhsa, 2024-03-24:
I think something like this, you might have to enable "Hidden Items" to be visible on your File Explorer.

C:\Users\**Your User Name**\AppData\Local\LiveSPICE\LiveSPICE_Url_w5ye0vm5hlaotc22ot05vljtu3mbfnzi\1.0.0.0

Then inside the last folder you should find the user.config file.

Morophla, 2024-03-25:
Ok! Got it! Thank you very much!

movleax, 2024-08-27:
I was experiencing this same issue.

It seems there is a problem with LiveSPICE->DockingManagerExtensions.cs->SaveLayout(...)

On line 26, "config = new StringBuilder();" is being reassigned and perhaps causing issues with the save state of the xml.
Commenting out this line resolved this issue.

I'm not versed at the AvalonDock, but I'm unsure why lines 20-30 are needed. Infact commented out these seemed to save the layout state just fine...
That reference to http://avalondock.codeplex.com/discussions/400644 doesn't turn up anything, nor does it on internet archive.

```
public static string SaveLayout(this DockingManager This)
{
    var serializer = new XmlLayoutSerializer(This);
    StringBuilder config = new StringBuilder();
    using (XmlWriter writer = XmlWriter.Create(config))
    {
        serializer.Serialize(writer);
    }

    // http://avalondock.codeplex.com/discussions/400644
    //XmlDocument configDoc = new XmlDocument();
    //configDoc.Load(XmlReader.Create(new System.IO.StringReader(config.ToString())));
    //XmlNodeList projectNodes = configDoc.GetElementsByTagName("LayoutDocument");
    //for (int i = projectNodes.Count - 1; i > -1; i--)
    //    projectNodes[i].ParentNode.RemoveChild(projectNodes[i]);

    ////config = new StringBuilder();
    //using (XmlWriter writer = XmlWriter.Create(config))
    //{
    //    configDoc.WriteTo(writer);
    //}
    return config.ToString();
}
```

I've never contributed to another github repo. Seems like "outsiders" wouldn't have access.
Maybe I'll fork and fix this.