dsharlet/LiveSPICE #202: Mac Port?

thouldcroft, 2023-08-16:
This project is something I have wanted for YEARS. Is there any way to port this to Mac, or does anyone have a rundown on how I could build it from source on Mac?

qwag, 2023-08-31:
im desperate for this on Mac too, something ive wanted for quite some time.

evan-olcott, 2024-03-10:
I'd like to be involved in this project - I've got lots of years of Mac/iOS audio dev under my belt, and this seems like a holy grail of sorts.

If I am understanding the code/docs correctly, then there's an analysis phase before the realtime phase, which sort of "compiles" the DSP applied later, and this compilation phase is done with something called Linq, which not only doesn't seem to have a kind of macOS counterpart, but also doesn't have any further development happening.

If this is the case, what kind of system could we use on macOS to simulate this? Is it worth re-writing `ComputerAlgebra` into something more cross-platform friendly, like C++ or Rust? And what would it mean to "compile" into some other DSP processing without something like Linq?

I love this idea, I think it's a really great tool for musicians/engineers, but to bring it to the Mac means understanding the math a bit more and how we can port it... Dillon, would you be willing to help us out?

dsharlet, 2024-03-11:
Thank you for the interest, I'm happy to help how I can of course.

Maybe just running the existing .Net code on Mac via .NET Core is an option? Looking at the current .NET downloads, there are quite a few MacOS options, including a current runtime version for MacOS: https://dotnet.microsoft.com/en-us/download/dotnet/8.0

It's not impossible to rewrite everything in e.g. C++, but it is of course a pretty big task. I generally like C++ (most of my other GItHub repos are C++), but writing something like a computer algebra system seemed easier in C#. We leverage LINQ in particular to compile programs at runtime. This is also possible in C++, but it requires bringing in a dependency like LLVM, or using a lighter weight runtime code generator (but we'd then have to implement a lot of compiler optimizations ourselves).

I'm not opposed to a complete rewrite, but it really is a pretty big undertaking. There are things I'd do differently if starting over today, but also some key ideas that I think are essential to making LiveSPICE what it is (capable of simulating interesting circuits in real time). 

Anyways, this is not something I can realistically just write down a how-to for, but if you have any particular questions you'd like me to expand on, please ask. 

BTW: I've realized writing this that there are at least 2 different issues I've written similar comments on. I'm going to look at consolidating all of these into a "discussion". If this issue disappears, that's what happened to it.

dsharlet, 2024-03-11:
As mentioned above, I'd like to consolidate cross platform porting discussions here: https://github.com/dsharlet/LiveSPICE/discussions/222

closed 2024-03-11 by dsharlet