dsharlet/LiveSPICE #204: Inquiry about Valid Voltage Source Expressions

standard-librarian, 2023-10-27:
I am seeking information about the valid expressions for voltage sources.
 While I'm aware that expressions like `Sin[6.28 * t] V` are accepted, I am interested in knowing if the software also supports other waveform types such as:

- Cosine
- Triangle
- Pulse

If these additional waveforms are not currently supported, I would like to inquire about the possibility of contributing to add support for them.

Thank you for your assistance.

dsharlet, 2023-10-27:
Unfortunately I think the best "documentation" for this is this bit of code that shows the functions that are available: https://github.com/dsharlet/ComputerAlgebra/blob/master/ComputerAlgebra/Namespace/Global/GlobalNamespace.cs

There's also these functions and constants, which are circuit/time-simulation specific: https://github.com/dsharlet/LiveSPICE/blob/master/Circuit/Simulation/TransientSolution.cs#L79-L86

So to answer your specific questions:
- Cosine is supported: `Cos` exists
- The delta (`d`) or step (`u`) functions *might* be what you mean by pulse?

For triangle/square waves and the like, they could be implemented like this:
- Half-triangle: `Mod[t*K]`
- Full triangle: `Abs[Mod[t*K] - 0.5]*2 - 0.5`
- Square: `Sign[Mod[t*K] - 0.5]`

We don't actually have the `Mod[x]` function today :) It can be implemented as `x - Floor[x]`. It would be a good idea to add such a function, alongside these: https://github.com/dsharlet/ComputerAlgebra/blob/master/ComputerAlgebra/Namespace/Global/GlobalNamespace.cs#L75

In all the above, `K` is a constant that determines the frequency. `t` is in seconds, so `K` has units `Hz` or cycles per second.

Federerer, 2023-10-27:
I used 'Sign[Sin[t]]' for the square wave.

standard-librarian, 2023-10-27:
@dsharlet  @Federerer  Thank you for your input!

I appreciate the suggestion of using `Sign[Sin[t]]` for creating a square wave. However, I've encountered a slight issue with this approach as it yields a value of 0 when the input is 0. To address this, I'm currently working with the expression `Ceiling[t*K - Floor[t*K] - (1 - d)]`, where `d` represents the duty cycle, and `K` is a constant defining the frequency.

Thank you for your help! Open to more.

standard-librarian, 2023-10-27:
Is it possible for me to open an issue on the [ComputerAlgebra repository](https://github.com/dsharlet/ComputerAlgebra) to propose the addition of more functions that could be beneficial for LiveSpice users like myself?

dsharlet, 2023-10-27:
Of course. If for some reason you don't have access, let me know.

I'm already thinking I should add `Mod[x] = x - Floor[x]` as a helper for the kinds of functions we discussed above. Not having that already is really an oversight. If you have other ideas I'm happy to hear them. For things that are more "circuit simulation specific" it might make sense to put them here (in the LiveSPICE repo) instead, and maybe I should figure out a better way of defining such things than hacking them into the simulation namespace like it does now with `d[x]` and `u[x]`.

dsharlet, 2023-10-27:
>  appreciate the suggestion of using Sign[Sin[t]] for creating a square wave. However, I've encountered a slight issue with this approach as it yields a value between the max value and the min value (0 between 1 and -1 or 0.5 between 0 and 1).

Another workaround to this would be to add a small offset to `t`, e.g. `T / 2` (`T` is the timestep of the simulation). But if your frequency isn't a rational factor of the simulation frequency, it'll just be random chance if you avoid hitting a zero (probably very rare but still).

This is all especially weird because it is floating point arithmetic, and the numbers get bigger (and less precise) as the simulation goes on. I don't think this happens at a scale that actually matters though.

I think your definition of a square wave is probably the best one proposed so far.

standard-librarian, 2023-10-28:
Awesome! I will open another issue for that to keep everything organized for people if they needed to ask about these things in the future!

closed 2023-10-28 by standard-librarian

mohamed-sallam, 2024-07-02:
@dsharlet 
I am trying to add Mod[x, y] function but it give me this
```
at System.Collections.Generic.Dictionary`2.FindValue(TKey key)
   at System.Collections.Generic.Dictionary`2.TryGetValue(TKey key, TValue& value)
   at ComputerAlgebra.LinqCompiler.CodeGen.LookUp(Expression Expr) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CodeGen.cs:line 130
   at ComputerAlgebra.LinqCompiler.CompileExpression.Visit(Expression E) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExpression.cs:line 20
   at ComputerAlgebra.LinqCompiler.CodeGen.Compile(Expression Expr) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CodeGen.cs:line 162
   at ComputerAlgebra.LinqCompiler.CompileExtension.CodeGenFunction(Function f, Module Module) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExtension.cs:line 23
   at ComputerAlgebra.LinqCompiler.CompileExtension.Compile(Function This, Module Module) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExtension.cs:line 33
   at ComputerAlgebra.LinqCompiler.Module.Compile(Function f, Type[] ArgTypes) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/Module.cs:line 112
   at ComputerAlgebra.LinqCompiler.CompileExpression.VisitCall(Call C) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExpression.cs:line 114
   at ComputerAlgebra.ExpressionVisitor`1.Visit(Expression E) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/Expression/Visitors/Visitor.cs:line 27
   at ComputerAlgebra.LinqCompiler.CompileExpression.Visit(Expression E) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExpression.cs:line 24
   at ComputerAlgebra.LinqCompiler.CompileExpression.VisitProduct(Product M) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExpression.cs:line 44
   at ComputerAlgebra.ExpressionVisitor`1.Visit(Expression E) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/Expression/Visitors/Visitor.cs:line 25
   at ComputerAlgebra.LinqCompiler.CompileExpression.Visit(Expression E) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExpression.cs:line 24
   at ComputerAlgebra.LinqCompiler.CompileExpression.VisitSum(Sum A) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExpression.cs:line 29
   at ComputerAlgebra.ExpressionVisitor`1.Visit(Expression E) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/Expression/Visitors/Visitor.cs:line 24
   at ComputerAlgebra.LinqCompiler.CompileExpression.Visit(Expression E) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExpression.cs:line 24
   at ComputerAlgebra.LinqCompiler.CompileExpression.VisitProduct(Product M) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExpression.cs:line 46
   at ComputerAlgebra.ExpressionVisitor`1.Visit(Expression E) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/Expression/Visitors/Visitor.cs:line 25
   at ComputerAlgebra.LinqCompiler.CompileExpression.Visit(Expression E) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExpression.cs:line 24
   at ComputerAlgebra.LinqCompiler.CompileExpression.<VisitCall>b__8_0(Expression i) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExpression.cs:line 113
   at System.Linq.Enumerable.SelectListIterator`2.ToArray()
   at ComputerAlgebra.LinqCompiler.CompileExpression.VisitCall(Call C) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExpression.cs:line 113
   at ComputerAlgebra.ExpressionVisitor`1.Visit(Expression E) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/Expression/Visitors/Visitor.cs:line 27
   at ComputerAlgebra.LinqCompiler.CompileExpression.Visit(Expression E) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExpression.cs:line 24
   at ComputerAlgebra.LinqCompiler.CompileExpression.VisitSum(Sum A) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExpression.cs:line 29
   at ComputerAlgebra.ExpressionVisitor`1.Visit(Expression E) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/Expression/Visitors/Visitor.cs:line 24
   at ComputerAlgebra.LinqCompiler.CompileExpression.Visit(Expression E) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CompileExpression.cs:line 24
   at ComputerAlgebra.LinqCompiler.CodeGen.Compile(Expression Expr) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CodeGen.cs:line 162
   at ComputerAlgebra.LinqCompiler.CodeGen.DeclInit(Scope Scope, Expression Expr, Expression Init) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CodeGen.cs:line 289
   at ComputerAlgebra.LinqCompiler.CodeGen.DeclInit(Expression Expr, Expression Init) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CodeGen.cs:line 297
   at Circuit.Simulation.<>c__DisplayClass44_1.<DefineProcess>b__5() in XXX/LiveSpiceApi/Circuit/Simulation/Simulation.cs:line 314
   at ComputerAlgebra.LinqCompiler.CodeGen.<>c__DisplayClass71_0.<DoWhile>b__0(Expression x, Expression y) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CodeGen.cs:line 471
   at ComputerAlgebra.LinqCompiler.CodeGen.DoWhile(Action`2 Body, Expression Condition) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CodeGen.cs:line 459
[... cut ...]
   at ComputerAlgebra.LinqCompiler.CodeGen.For(Action Init, Expression Condition, Action Step, Action Body) in XXX/LiveSpiceApi/ComputerAlgebra/ComputerAlgebra/LinqCompiler/CodeGen.cs:line 368
   at Circuit.Simulation.DefineProcess() in XXX/LiveSpiceApi/Circuit/Simulation/Simulation.cs:line 263
```

I added these lines:
```c#
// GlobalNamespace.cs
//...
        public static Expression Mod(Constant x, Constant y) { return Real.Mod(x, y); }
//...
```
and
```c#
// Real.cs 
//...
        public static Real Mod(Real x, Real y) { return new Real((double)x % (double)y); }
//...
```
Are there any other place that need me to declare Mod[x,y] in?

dsharlet, 2024-07-03:
I suggest searching the codebase for `ArcCoth` (or any other function we already have, but I think this one minimizes spurious results), which will find all of the places support is needed:

- `Expression\Call.cs`
- `Extensions\Differentiate.cs`. I think this should be `new SubstituteTransform("D[Mod[u, a], x]", "D[u, x]"),`, i.e. just treat it as u[x] and ignore the modulus. Pretty sloppy but probably what LiveSPICE wants (see also: `Sign[u]`).
- `LinqCompiler\StandardMath.cs`, I think this is the one you are missing in the error you got. *Note there are 2 in this file, double and float*.
- `Namespace\Global\GlobalNamespace.cs`, you got this one.
- `Util\Real.cs`, you got this one.

dsharlet, 2024-07-03:
For the `Real.Mod` implementation, `BigRational` has an `operator%` which avoids converting to double, which is really a pretty lossy operation.

mohamed-sallam, 2024-07-07:
@dsharlet , thank you very much for your help we will create PR, God willing, with Mod[] function and some other function like `pulsetrain` and ASP.net core API, if it will help.