dsharlet/LiveSPICE #206: Importing SPICE Macro Models into LiveSPICE

standard-librarian, 2023-10-28:
Description:

Inquiry about Importing SPICE Macro Models:
I'm interested in importing a specific SPICE macro model, the "AD8411A SPICE Macro Model" from Analog Devices, into LiveSPICE. This model has been made available by Analog Devices.

Reference to SPICE Macro Model:
The SPICE macro model I'm interested in can be found here: [AD8411A SPICE Macro Model](https://www.analog.com/media/en/simulation-models/spice-models/ad8411a.cir)

Question:
Considering my limited knowledge of SPICE simulations, I would appreciate it if I could know whether LiveSPICE supports the import of external SPICE macro models or sub-circuits. If it does, I'd like to understand how I can import the "AD8411A SPICE Macro Model" or similar models into LiveSPICE.

Additional Information:
I've tried to take (.cir or .spice.txt) file, add them to the Components folder. It gave me the devices with the `.models` lines rather than the single macro model.

dsharlet, 2023-10-31:
Supporting more SPICE models is something I've wanted to do (#22), but haven't gotten to in part because it is a long tail of challenging cases to support. It's not something that can be done "completely" I don't think.

Another issue with it is that I think even really simple SPICE models would push the limits of LiveSPICE. I think big components like this need approximating to make practical for real time simulation. It looks like this example has 20+ non-linear component. We can maybe simulate this model by itself, but not much else at the same time. This is especially unfortunate because looking at the datasheet, it looks like at least a bunch of the non-linear components are emulating an op-amp. So to make this component more practical in LiveSPICE, we'd want to model that op-amp at a higher level, all of this is not something readily automated.

standard-librarian, 2023-10-31:
I came across the concept of Lookup Tables (LUTs) and their potential for enabling realtime simulations. Although I lack the necessary domain knowledge, I’m intrigued by the possibility of integrating LUTs into liveSPICE.

dsharlet, 2023-10-31:
It's possible there are interesting ways to approximate parts of circuits, but there is a huge set of possible solutions to consider, and I think any one of them is a serious research project, unless we can find a specific technique that is well documented enough to implement.

I'm a bit skeptical of LUTs specifically for two reasons:
- The dimensionality gets crazy quickly. For example, AD8411A looks like it would have 5-7 pins. With that many dimensions, it's hard to have a detailed LUT.
- Capacitors introduce state. It's not clear to me how to handle this with a LUT, which is stateless. Maybe by adding more dimensions to the LUT, but that seems like a step in the wrong direction. Perhaps it's surprising, but there are a few capacitors in the AD8411A spice model.

If I were to start working on this today, I'd probably pursue some tiny recurrent MLP like neural network. The recurrence could hopefully handle the statefulness of capacitors. But as I mentioned above, this is a serious research project, maybe there is some existing research that can be implemented?