dsharlet/LiveSPICE #207: Request for Backend Rewrite in a Cross-Compatible Language for liveSPICE

standard-librarian, 2023-10-31:
We’re looking into the possibility of rewriting the backend of liveSPICE in a different language to address the challenges related to using C# and .NET on Mac and Linux systems. The current backend’s reliance on C# and .NET poses limitations for cross-platform compatibility, hindering its accessibility and usability on non-Windows environments.

To enhance the accessibility and versatility of liveSPICE, we’re considering alternative languages such as C, Java, Python, Go, or Rust, which offer better cross-platform support and are more conducive for Mac and Linux users.

M y teammate and I are open to exploring potential contributions to this migration effort, considering the rewiring of the liveSPICE backend for improved cross-compatibility. If the project aligns with our capabilities and goals, we would consider actively engaging in the migration process, should it be a feasible and practical endeavor.

dsharlet, 2023-10-31:
There are a few things that C# gives us that are hard to replicate in other languages:
- I like making GUIs in C# more so than other languages. LiveSPICE (at least the GUI application) is a pretty complicated GUI.
- We use the LINQ expression mechanism to provide us with a really nice easy to use JIT compiler. This isn't impossible in other languages, but it usually means pulling in a pretty big dependency (e.g. LLVM), or doing a lot more of the low level optimizations ourselves.

That said: I think .net core *is* supported on Mac and Linux? Just googling around, people seem to think that .net core (3.1 or later) on Linux is a pretty good environment and performs well (this was not true prior to 3.1).

Federerer, 2023-10-31:
It is. The only thing that is not supported is WPF. I once considered
migrating the frontend to Avalonia, which is cross-platform, but that's a
lot of work.