dsharlet/LiveSPICE #208: How to add models?

dustinlacewell, 2023-11-26:
Hello, I'm fairly new to all of this and could use some guidance. I'm trying to learn analog synthesis and I need a Inverting Schmitt Trigger. Can anyone advise me how to get such a component into LiveSPICE? 🙏

dustinlacewell, 2023-11-26:
I've tried adding a few `.lib` files from around the internet, but in LiveSPICE they all just show up as this:

[image]

dsharlet, 2023-12-03:
LiveSPICE currently only supports importing basic component models. I'm guessing that in this case, the thing you are looking for is a `.subckt` directive, which isn't supported.

It might be a bug that the thing you are looking for shows up as a diode though, if you share your `.lib` file, I can take a look at that.

dustinlacewell, 2023-12-05:
Here is the .lib for the CD40106BE Hex Inverting Schmitt Trigger:

https://gist.github.com/dustinlacewell/191a8329ea86771def321daa4b340b86