dsharlet/LiveSPICE #209: Suggestion: Middle-click panning

JamesC01, 2023-11-29:
Right now it's very inconvenient to pan in the schematic view, you have to use the scrollbars as far as I can tell. It would be really nice to middle-click drag to pan around. I'm not sure how to implement it myself, so I can only make a suggestion, but I think it would be quite useful.

dsharlet, 2023-11-30:
Thanks for the feedback. I'll give a few things a try along these lines. I'd like to use an existing convention in similar software if possible, did you get the idea for middle click + drag from some other software?

I think I scroll by zooming out and zooming back in where I want to go (with the mouse wheel), so I've been subconsciously working around this issue!

JamesC01, 2023-11-30:
> Thanks for the feedback. I'll give a few things a try along these lines. I'd like to use an existing convention in similar software if possible, did you get the idea for middle click + drag from [...]

Middle-click drag is the usual convention for panning. I'm not sure if it is in most SPICE programs, but falstad is an example, and also many photo-editing software, 3d modelling software (shift-middle-click drag). It just seems like a convention for panning. 

Also, while you're here, another small suggestion I have is a slightly more visible quick-edit for component values (resistance, capacitance etc.) when you press F2. It would be nice if the label showing the value next to the component could update in real time, and maybe have the text cursor there to show that you're editing it. So when you have a component selected and you press F2, it would highlight the current label, showing that you're about to overwrite it, like when you click-drag select text and it turns blue, and then when you start typing, the label right next to the component could update as you're typing, instead of only after you press enter.

dsharlet, 2023-12-03:
I've just pushed a commit that adds middle click + drag to scroll.

As for your second suggestion, I've made selections a bit more visible (the subtlety of that has caught me a few times), but I think doing much more than that would be pretty challenging. The UI doesn't actually know anything about how the label is drawn, it would require quite a lot of plumbing of stuff to make that work. Updating the value in real time also has problems, due to a bunch of validation that will fail (e.g. while you type an expression with parentheses, the intermediate edits are not valid). I think supporting this would require another layer of UI logic that just doesn't exist right now.

Hopefully the selection being more visible at least partially addresses the issue.

closed 2023-12-03 by dsharlet

JamesC01, 2023-12-03:
> I've just pushed a commit that adds middle click + drag to scroll.
>
> As for your second suggestion, I've made selections a bit more visible (the subtlety of that has caught me a few times), but I [...]

I just tried it out, and the scrolling is smooth, but you've got it inverted from what most software uses. When you middle-click drag to the right, usually it moves to the left, same as if you were dragging on a touch-screen. Same with up and down. It's still a lot nicer that having to use the scroll bars, though. Thanks!

Now that you've added the majority of the code, I can understand it and I figured out how to make it scroll like it does in many programs. It usually feels like you're dragging on a touch screen, so I inverted the values, and removed the zoom multiplier, because the zoom multiplier makes the screen drag out-of-sync with the mouse, which most programs don't do:

`scroll.ScrollToHorizontalOffset(scroll.HorizontalOffset - dx.X);`
`scroll.ScrollToVerticalOffset(scroll.VerticalOffset - dx.Y);`

Now it feels exactly like it does in most programs, just like a touch screen.