dsharlet/LiveSPICE #213: Simulation Diverged...

AdmiralBongo, 2024-01-17:
[image]
[VH140C Channel A.schx.txt](https://github.com/dsharlet/LiveSPICE/files/13969710/VH140C.Channel.A.schx.txt)

I tried to simulate the Ampeg VH140C lead channel and without any luck. Moment I load it up, I get an error. Could be the unconnected opamp terminals (they're that way in the official schematics on the Ampeg site and in every pedal VH140C clone schematic I've found), could be the opamps themselves (or rather load they put because of how many of them there are), but swapping 4558's for the "ideal opamps" is pretty much useless, as, while it stops the errors, all I get is pure silence.

dsharlet, 2024-01-27:
This simulation does work for me as-is when I downloaded it. It produces an output which is not close to being zero mean, which might be a problem, but it seems to be working even despite that.

One issue is that this circuit is quite large, and so I needed to turn down "Oversample" to get it to run in real time. It's possible that if it doesn't run in real time, the "hiccups" introduced by failing to keep up with real time introduce large transients, which could cause the simulation to diverge (but not at 0s).

dsharlet, 2024-01-27:
I'm looking at doing a release now, and I see that since the last release, I made a bunch of changes to the way we understand wires and connectivity, It's possible that unconnected terminals are behaving better now, which could be why the original schematic works for me.

dsharlet, 2024-01-27:
I actually went and tried the previous release, and it is working there too, so there aren't any recent fixes that are affecting the issue, at least for me...

AdmiralBongo, 2024-01-29:
Weird. Anyhow, here's an updated schematic, which is closer to the original amp, having all opamps powered with two +/-15V rails. Just in case, I added 1N746 and 1N754 models, but I've tried the included generic zener diodes as well, with the same result. And the conclusion I've come to is that issue is at the distortion stage, as the tonestack works flawlessly when connected directly to the input.
[VH140C Channel A 3.schx.txt](https://github.com/dsharlet/LiveSPICE/files/14089337/VH140C.Channel.A.3.schx.txt)

AdmiralBongo, 2024-01-30:
I've managed to get it working in the release 15. However, it's still barely usable in the plugin, for it starts glitching out immediately even with the largest buffer size. This doesn't even allow to tweak the settings or use it during the mixing properly. It's basically "rendering-only". Guess I'll have to wait for stability/efficiency upgrades.