dsharlet/LiveSPICE #214: VST returns Runtime Error in REAPER

Suzuki-Re, 2024-01-24:
Hi,

I just installed [Preview 14](https://github.com/dsharlet/LiveSPICE/releases/tag/v0.14) and [Microsoft .NET 6.0 runtime](https://dotnet.microsoft.com/en-us/download/dotnet/thank-you/runtime-desktop-6.0.16-windows-x64-installer) as instructed.

The standalone version works fine, but VST3 gives me an error when opening it in REAPER v7.09.
[image]

How can I address the issue? I tried reinstalling the plugin and runtime, and restarting PC, but no luck. Do they conflict?
```
.NET runtimes installed:
  Microsoft.AspNetCore.App 7.0.15 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 6.0.16 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 7.0.0 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 7.0.15 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 6.0.16 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 7.0.0 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 7.0.15 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
```
[image]

Thanks.

Federerer, 2024-01-24:
This might be the framework version problem that we've already solved in AudioPlugSharp some time ago. Seems like `"rollForward": "LatestMajor"` was set in runtime config in version 0.5.6 and LiveSPICE still uses 0.5.0. So we should probably do an update. @mikeoliphant

Federerer, 2024-01-24:
Looks like we already have those changes in master, so maybe we need a new release @dsharlet?

Suzuki-Re, 2024-01-27:
Thank you for the reply! It would be awesome if @dsharlet can release a new one.

dsharlet, 2024-01-27:
I've done a new release. I'm assuming it will fix this issue, so I'm going to close it, please re-open if you still see this.

Thank you for reporting the issue!

closed 2024-01-27 by dsharlet

Suzuki-Re, 2024-01-27:
Thank you so much! It certainly fixed the issue!