dsharlet/LiveSPICE #223: Power Issue

synack-dev, 2024-03-12:
I'm just currently playing around with LiveSPICE and built a simple opamp circuit with a pot for gain and voltage divider to supply Vcc+ and Vcc- to the Opamp from a 9V supply. 
Oddly what happens is that i get clipping on the negative side of the signal pretty quickly, but never clips on the Positive side. I've seen voltages >+100v out of the op amp from this circuit which seems frankly to be a bug. (opamp voltage is 2.5v Vcc+ to .5v Vcc-)
[image]
[OpAmp.schx.txt](https://github.com/dsharlet/LiveSPICE/files/14576011/OpAmp.schx.txt)
Attached is a screencap of the circuit as well as the schx file. 
FYI this circuit works in Falstad as expected, so I presume it's a valid circuit and that I haven't done anything "crazy"

synack-dev, 2024-03-12:
Equiv circuit in Falstad is stable with clipping
[image]

dsharlet, 2024-03-13:
Thank you for the nice reproducer. I couldn't actually download your example (github link seems dead...) but I was able to reproduce the circuit easily enough and see the same thing you see.

I'd be a little surprised if the issue was actually some kind of simulation mismatch. I'm finding it a little tricky to inspect the circuits for similarity.

I did find one fairly substantial difference (capacitors in the grounding circuit are 47 uF vs. 240 uF), but that didn't actually affect the issue at hand. 

There's another *possible* difference: is the factor of 0.1 intended in the voltage source function? I can't see if that exists in the Falstad circuit too.

dsharlet, 2024-03-13:
I'm trying to replicate Falstad's result in LTspice, and I haven't been able to do that yet either. My results are very sensitive to the particular op-amp model I'm using. I tried replicating LiveSPICE's model in LTspice to the best of my ability, and I get something similar to what you describe with clipping on only one side:

[image]

LiveSPICE's op-amp model is documented as the "Op-Amp Model.schx" example.

Can you try replicating the circuit you want in LTspice and maybe we can then track down the right op-amp model?

synack-dev, 2024-03-13:
Same circuit in simulation. Gain all the way up. ( I reduced the speaker gain as it's a terrible sound lol)
[image]

synack-dev, 2024-03-13:
I think I might know what the issue could be. If I replace the Vcc+ with a 2.5V rail it behaves as expected.
[image]
I think that the voltages coming out from the opamp are being added to the voltage divider circuit which increases the VCC+ and so on... 
So I have a work around for the time being

synack-dev, 2024-03-13:
I thought maybe the output was biasing the voltage divider network via the Gnd connection. I placed a diode to prevent the extra voltage from GND to go back up to Vcc+. Still getting the extra voltage on Vcc+ from the output somehow. Is it possible there is a voltage leak in the simulation code from Output to Vcc+ on the Opamp?
[image]
[image]

dsharlet, 2024-03-13:
Right, I should have paid more attention to that out of range voltage issue. I hadn't seen that in my repro attempts. I see now, if I turn R4 up to near 1, the voltage of the output can get very large. 

I'm almost certain the problem is in the op-amp model. It uses a voltage controlled current source that can generate arbitrarily large currents. There's probably something missing from the model that limits it. I'm trying to figure out LTspice's UniversalOpAmp2's implementation to see if there's something similar we can do.

synack-dev, 2024-03-13:
I've looked at the LiveSpice code. Well done!
I'm having a hard time interpreting the analyze functions, but one thing I noticed was the use of "voltage source" object for the Vcc and Vee. I'm not sure that is correct, They are not really a voltage source. I could be interpreting it wrong for sure, but it seem to me that Voltage Source would be more like a battery or something that OUTPUTS voltage. 

It might appear to work fine when attached directly to nothing but a voltage rail. When added to a circuit with other voltages, it could become additive and display the issues i've encountered. The negative rail is 0 so 0+0 = 0 and therefore there would be no change. However on the positive side, with capacitors and an iterative process, the voltage would keep adding up as i'm seeing. 
I think the Vcc and Vee should be "in" inputs and treated much like the P and N with resistance between Vcc and Vee.  P and N should never exceed Vcc and Vee. Later you can add properties for rail to rail losses for the real opamps.

dsharlet, 2024-03-13:
The voltage sources attached to Vcc and Vee are "only" adding (or subtracting) 2V from the actual supply voltages. TBH, I can't remember why I put those there. I was implementing some model I found somewhere, I wish I saved a link to it...

I think it (these voltage sources specifically) should be OK even if you don't have a fixed voltage attached. I suspect the problem is one (or both) of these issues:
- The 2V voltage sources in the model might do weird things when Vcc - Vee is <= 4 (these sources will not allow this, or maybe it makes the diodes behave strangely).
- The current source might be able to get arbitrarily large and there should be some kind of limit on it.

I might just attempt to re-implement the op-amp model using the LTspice universal op amp model as a guide. I think the LTspice model maybe accomplishes the same thing these voltage sources are attempting to do with some resistors that act like voltage dividers, but I'm not sure. That seems like a much better approach than just forcing these voltage sources in.

I'm kind of amazed that this model has been in place for so long, and it's taken this long for such a basic issue to appear!

dsharlet, 2024-03-13:
Oh, I did save the link I was thinking of! The link is broken because they moved the file, but the new location is here: https://qucs.sourceforge.net/docs/tutorial/opamp.pdf

On page 31, similar voltage sources appear. However, it looks quite different than I remember. This looks more like an example test circuit, rather than a general model, which makes sense (the 2V offset doesn't make sense in general).

synack-dev, 2024-03-13:
The output *does* have a voltage limit in some opamps of Vcc - 2v and Vee +2 volts. The LM324 my circuit is based on is Vee to Vcc-1.5v. Hence the 2.5v Vcc and .5 V Gnd potential, so that the output would be centered about 0.5V, swinging between 0 (Vee) to 1V (Vcc-1.5) mark.
For a "generic" opamp it's fine to say Vee+2 to Vcc-2 as the common mode output range. Output voltage should be clamped to those values when saturated.
If I had used the rails instead of a voltage divider circuit, or had lower gains it probably wouldn't have appeared for me either.
I was purposefully driving the opamp hard to see what output I would get, both in the scope and actual audio. 

That section of the PDF is a bit hard to read as the figures don't really line up with the text. In some cases the figure is pages after the text.