dsharlet/LiveSPICE #224: Missing Non Generic Components

ncblue21, 2024-03-21:
Hi, First Livespice is awesome! I've been using v13 over a year and have created many guitar effects and run it in vsthost.exe with other plugins. Great for designing new effects, mostly overdrives. I believe it gives a better tonal response than many plugins. At least the free plugins. 

Issue: I upgraded to v15 and the non-generic components are now gone. So no numbered diodes, transistors, opamps. I tried to move the old component files over, no luck. I can work around it by copying components from another circuit or inputting properties from LTspice.

Thanks for the great work!

dsharlet, 2024-03-22:
Thank you for the nice comments, it's great to hear your positive feedback!

Sorry to hear about the missing components. Unfortunately I haven't been able to reproduce it by uninstalling and reinstalling LiveSPICE. 

Do you see these files in your LiveSPICE installation? This is where those components come from:
[image]

LiveSPICE looks for this folder in the same location as the executable itself.

I know this is an annoying suggestion, but if the above doesn't offer any clues, maybe just uninstalling and reinstalling is worth a try?

ncblue21, 2024-03-25:
I did have that folder with the component xml files, so I backed up my files, uninstalled, deleted any folders and reinstalled v15. Success! Thanks and keep up the good work.

RobertKeyser, 2024-04-04:
> Hi, First Livespice is awesome! I've been using v13 over a year and have created many guitar effects and run it in vsthost.exe with other plugins. Great for designing new effects, mostly overdrives. [...]

I had this same problem with the upgrade on Windows 10 (made me update the .Net desktop runtime). What is the correct upgrade path? uninstall first?

ncblue21, 2024-04-05:
I backed up my files then uninstalled. Then made sure there was no leftover files and folders and even cleaned the registry.