dsharlet/LiveSPICE #227: Inno Setup Compiler Abort - LiveSPICE\bin Missing

student237, 2024-04-30:
Hello, I attempted to build the LiveSPICESetup.exe file with Inno Setup Compiler, but it was aborted due to unavailable source files. I may be missing something somewhere, but I don't see a _LiveSPICE\bin_ folder anywhere in the repository.  Here is a screenshot of the error. Thank you! 

[image]

dsharlet, 2024-04-30:
These files are generated by building LiveSPICE in visual studio, have you
done that before running the setup compiler?

student237, 2024-04-30:
No, my apologies, I haven't. I installed Microsoft .NET 6.0 runtime and ASIO4ALL, then followed the steps at https://livespice.org/#getting-started. I'll jump back in and review everything again as I had a feeling I missed a step somewhere. Thank you for getting back to me so quickly!

dsharlet, 2024-04-30:
If you’re just trying to use LiveSPICE, you don’t need to build the
installer or run visual studio, that’s how the installer gets created. You
probably just want to download an existing release from here:
https://github.com/dsharlet/LiveSPICE/releases

closed 2024-12-03 by dsharlet