dsharlet/LiveSPICE #234: I cant Use the arrow keys to rotate and flip components while adding them to the schematic.

hal0801, 2024-05-27:
Nice to meet you,
arrow keys to rotate and flip components while adding them to the schematic.
but only the cross keys are displayed and I can only select a range.
I tried holding down the shift key, alt key, etc. but nothing changed.
Sorry for the newbie question, but I'm really having trouble. Thank you in advance for your help.

edoardogiacomello, 2024-05-27:
Are you using the ARROW keys or the NumPad? For me it works correctly using arrow keys when the item is selected (blue border).

hal0801, 2024-05-27:
It is done!
Thank you so much!

> Are you using the ARROW keys or the NumPad? For me it works correctly using arrow keys when the item is selected (blue border).

It is done!
Thank you so much!