dsharlet/LiveSPICE #244: Can't use multiple instances of LiveSPICE vst in Reaper DAW

joshdevvv, 2024-12-22:
Did a cursory look to see if this was an existing issue, I apologize if it is. When I add another instances of the VST, it mirrors the first one, and vice versa - even if they are completely separate tracks. 

I remember seeing the Rohbemusic linked on the homepage and seeing he was using a simulation of a pedal in front of a simulated jcm800

RickOmegaStation, 2025-01-09:
I got the same problem, I'm using VST3 on Reaper, it mirrors the first one and all the audio starts stuttering. I had no problem using two or more instances in the passt, probably it was with the previous version (V0.14).

AdmiralBongo, 2025-01-20:
Got the same thing happening.

MARKTHERENCE, 2025-03-10:
REAPER user here. I had the same issue as well while trying to run a simulation of a Martin Golub Lead (Green Day "Dookie") Marshall mod schematic with the preamp and poweramp sections separate from each other. You need to set the VST to run as a dedicated process.
`Add FX > Right-click on LiveSPICE > Run as > Dedicated process.`

joshdevvv, 2025-03-10:
> REAPER user here. I had the same issue as well while trying to run a simulation of a Martin Golub Lead (Green Day "Dookie") Marshall mod schematic with the preamp and poweramp sections separate from [...]

Thanks for sharing the fix!

RickOmegaStation, 2025-03-10:
> REAPER user here. I had the same issue as well while trying to run a simulation of a Martin Golub Lead (Green Day "Dookie") Marshall mod schematic with the preamp and poweramp sections separate from [...]

It works like a charm. Thank you very much! I gave up using it because of this issue, now I will try to make some more circuits

closed 2025-06-02 by mikeoliphant