dsharlet/LiveSPICE #248: VST not recognized by DAW

ludwigparker, 2025-01-15:
I recently had to factory reset my computer, and as a result had to reinstall all of my audio softwares. Everything except for the VST version of LiveSPICE works as usual, and it's really confusing. 
My DAW is Reaper. I have used this DAW for years, and also used LiveSPICE in conjunction with Reaper without issue. The standalone version of LiveSPICE works perfectly fine, but no matter how many times I reinstall both programs, change/update directories, revert to older versions of each program, it seems like Reaper absolutely refuses to recognize LiveSPICE as a VST. It's not even showing up in the plugin list, it's entirely nonexistant. As far as I can tell, Reaper reads every other plugin I've tried. 
I can use the standalone program if need be, but seeing as my background is more in practical circuit applications for guitar and bass gear, it would be hugely beneficial to be able to load circuits into a VST chain with other plugins such as overdrives, cabinet/IR loaders, et cetera. I don't know jack about programming so I am completely at a loss right now. 
I'm not home right now but if need be, I can upload some pictures of what's going on to help clarify.

ludwigparker, 2025-01-16:
I just reinstalled a prior version, v0.13 to see if the issue was with VST3. Lo and behold, Reaper now reads the VST version. The issue was with VST3, but with the improvements of later versions, it's limiting to only be able to use a 4+ year old version. Hope this gets fixed soon. Or, are there any ways to fix it on my end with DAW settings?

rossbalch, 2025-04-23:
Are you sure it's not Recognised by Reaper? If you search for LiveSpice does it show up? I had to add the "No Manufacturer" or it was "No Developer" category and then it shows up in that category for me.

alissaan1, 2025-06-11:
@ludwigparker I followed [this article](https://www.reddit.com/r/Reaper/comments/l9as4f/vst_plugins_failed_to_scan/) and it worked.

Basically, you just need to install a bunch of Visual C++ Redistributable packages.
An easy way to do it: https://www.techpowerup.com/download/visual-c-redistributable-runtime-package-all-in-one/

Hope it helps!